adafruit/circuitpython #3655: I2C device scanning

ronfischler, 2020-11-08:
I am looking to be able to hotplug an I2C device into my board. The normal method of scanning the entire I2C address space takes a bit too long for my liking and there does not seem to be a simple method to check for the presence of a single device. To get around this, I wrote a function to just scan for something at a specific address:

```
def I2C_checkDev(i2c, *args):
    for addr in args:
        if isinstance(addr, int) and (0 <= addr <= 0x7F):
            try:
                i2c._i2c._i2c_bus.read_byte(addr)
            except OSError:
                return False
    return True
```

Then assuming I have used busio.I2C() to assign i2c, I can scan for a single device:
I2C_checkDev(i2c, 0x40)
to check if this particular device is currently present.

Timing:
When using the existing i2c scan() method, it takes about 100 msecs to scan the bus for connected devices. When using the above code to check for the presence of a single device, I am usually coming in a bit over 1 msec. In my case, I am using this function on my RPi Zero board where I want to check periodically for an I2C device to appear. I plan to use a part such as the TI TCA4307 to support hot-swappable I2C.

I am wondering if such a method would be useful to others, if added to Adafruit's i2c.py?

ladyada, 2020-11-09:
if you just want to scan for a single address, try to create an I2CDevice with that address, if it succeeds, it means it found it 👍 
https://github.com/adafruit/Adafruit_CircuitPython_AHTx0/blob/master/adafruit_ahtx0.py#L66
https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/master/adafruit_bus_device/i2c_device.py#L163

ronfischler, 2020-11-09:
Thank you ladyada, that is a nice solution. Testing by device creation instead of a byte-read attempt makes sense. In either case the user simply has to catch the exception if the device is not connected. Additionally, calling I2CDevice(i2c, addr) certainly makes for more readable code than calling i2c._i2c._i2c_bus.read_byte(addr). My testing indicates I2CDevice() is also quite fast, so no problem there. I have much to learn.

closed 2020-11-09 by ronfischler

ladyada, 2020-11-09:
awesome :)