adafruit/circuitpython #3660: Esp32s2 saola wrover board add display by builting.

BradChan, 2020-11-09:
I try to Add new board from espressif_saola_1_wrover .Inorder to build the display module for the board, i modify the board.c and pins.c file. It builds and flashes successfully. But the circuitpython is not run. There is no CIRCUITPY appears.
[espressif_saola_1_wrover.zip](https://github.com/adafruit/circuitpython/files/5508068/espressif_saola_1_wrover.zip)

tannewt, 2020-11-09:
Are you adding a new board definition for a new board? Could you post a picture of your setup?

BradChan, 2020-11-10:
@tannewt 
[image]

tannewt, 2020-11-10:
@BradChan You don't need to compile a new board to get an external display working. You can initialize it with the Python library. Here is an example: https://github.com/adafruit/Adafruit_CircuitPython_ST7789/blob/master/examples/st7789_320x240_simpletest.py

closed 2020-11-10 by tannewt