adafruit/circuitpython #3663: FeatherS2 ESP32S2 wifi.radio.stop_scanning_networks() before scan bricks device

pvanallen, 2020-11-09:
On the FeatherS2, if code.py has wifi.radio.stop_scanning_networks() with no scan first, it causes the device to hang, stops showing up as CIRCUITPY, and there is no way to get the REPL. This is with MacOS and CircuitPython 6.0.0-rc.1. The board was purchased from Adafruit Nov 6, 2020.

See attached code:

[code-crash.py.zip](https://github.com/adafruit/circuitpython/files/5513427/code-crash.py.zip)

Restoring the S2 requires the following (must have esptool.py 3.0+ for this to work):

1. get into download mode and then `esptool.py --port /dev/tty.usbmodem01 erase_flash`
2. reinstall boot loader [according to this](https://feathers2.io/install_uf2.html)
3. reinstall the current CPY for ESP32

tannewt, 2020-11-09:
I've rephrased `brick` -> `hang`.

I believe safe mode can also help recover. You can enter it manually by pressing the boot button while circuitpython is starting. The RGB LED should be yellow during that period.

anecdata, 2020-11-09:
It may be trickier on FeatherS2 with the Dotstar, but looks like BruceS is on the case [https://discord.com/channels/327254708534116352/537365702651150357/775452077127237634](https://discord.com/channels/327254708534116352/537365702651150357/775452077127237634)

astrobokonon, 2020-11-09:
I wonder if this is another case of a flag (```self->scanning```) being in an inapplicable state like in #3580?  By calling ```stop_scanning_networks()``` you end up here

```c
void wifi_scannednetworks_deinit(wifi_scannednetworks_obj_t* self) {
    // if a scan is active, make sure and clean up the idf's buffer of results.
    if (self->scanning) {
        esp_wifi_scan_stop();
        if (wifi_scannednetworks_wait_for_scan(self)) {
            // Ignore the number of records since we're throwing them away.
            uint16_t number = 0;
            esp_wifi_scan_get_ap_records(&number, NULL);
            self->scanning = false;
        }
    }
    wifi_scannednetworks_done(self);
}
```

I can probably test myself in a few days (my ESP32S2 is tied up right this moment), but I'm wondering if ```esp_wifi_scan_stop()``` is really returning ```ESP_ERR_WIFI_NOT_INIT``` or ```ESP_ERR_WIFI_NOT_STARTED``` and that's hanging up the board.

tannewt, 2020-11-10:
@astrobokonon I think it's simpler. We simply need to return early if `current_scan` is NULL.

tannewt, 2020-11-10:
Or better yet, raise an error to point out the code is stopping out of order.

microDev1, 2021-02-27:
Fixed by #3819. Thanks!

closed 2021-02-27 by microDev1