adafruit/circuitpython #3669: Library does not work with Feather S2 and Circuit Python 6.x

jdoscher, 2020-11-10:
Using the guide from Adafruit here:
https://learn.adafruit.com/adafruit-eink-display-breakouts/grayscale-29-overview

The code returns the following error:
`code.py output:
Traceback (most recent call last):
  File "code.py", line 26, in <module>
  File "adafruit_il0373.py", line 184, in __init__
TypeError: extra keyword arguments given`

The offending code is this section, with line 26 the first line here:
`display = adafruit_il0373.IL0373(
    display_bus,
    width=296,
    height=128,
    rotation=270,
    black_bits_inverted=False,
    color_bits_inverted=False,
    grayscale=True,
    refresh_time=1,
)`

The full code is here:
`"""Simple test script for 2.9" 296x128 grayscale display.

Supported products:
  * Adafruit 2.9" Grayscale
    * https://www.adafruit.com/product/4777
  """

import time
import busio
import board
import displayio
import adafruit_il0373

displayio.release_displays()

# This pinout works on a Feather M4 and may need to be altered for other boards.
spi = busio.SPI(board.SCK, board.MOSI)  # Uses SCK and MOSI
epd_cs = board.D9
epd_dc = board.D10

display_bus = displayio.FourWire(
    spi, command=epd_dc, chip_select=epd_cs, baudrate=1000000
)
time.sleep(1)

display = adafruit_il0373.IL0373(
    display_bus,
    width=296,
    height=128,
    rotation=270,
    black_bits_inverted=False,
    color_bits_inverted=False,
    grayscale=True,
    refresh_time=1,
)

g = displayio.Group()

f = open("/display-ruler.bmp", "rb")

pic = displayio.OnDiskBitmap(f)
t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
g.append(t)

display.show(g)

display.refresh()

print("refreshed")

time.sleep(120)
`

dingramax, 2020-12-10:
I'm having the same issue with the feather m4 express

skieast, 2020-12-10:
I've been using a FeatherS2 with the Adafruit 2.9" Grayscale successfully using very similar code. 
My setup looks like this

```Python
# these are for the FeatherS2
epd_dc = board.IO3
epd_cs = board.IO1

displayio.release_displays()
display_bus = displayio.FourWire(board.SPI(), command=epd_dc,
                                 chip_select=epd_cs,
                                baudrate=1000000)
time.sleep(1)
display = adafruit_il0373.IL0373(
    display_bus,
    width=296,
    height=128,
    rotation=270,
    black_bits_inverted=False,
    color_bits_inverted=False,
    grayscale=True,
    refresh_time=1,
    seconds_per_frame=5
)
```
The code is basically the same as the Magtag Countdown code except for the pin changes to accomodate the FeatherS2

dhalbert, 2021-03-04:
I tested this with a FeatherS2, CircuitPython 6.2.0-beta.3, and the latest `adafruit_il0373.mpy`. There were no errors.

One thing that I have seen is that a stray copy of a library is left in `/` on `CIRCUITPY` instead of in `lib/`. `sys.path` is set so that `/` searched before `lib/`. Also, any `.py` version of a file takes precedence over a `.mpy` version, so if both were available, you might have gotten an older version of the library.

I will close this for now but please reopen if you can reproduce with the latest and with a clean filesystem.

closed 2021-03-04 by dhalbert