adafruit/circuitpython #3673: One wire issue on Pynt PyPortal D3 and D4 swapped - but not really...

wifijt, 2020-11-11:
I am using CPython 6.0 RC2 

I am trying to use a DS18B20 on a Pynt on D3 (yes, the resistor is installed) along with D4 as a digital input. However when I use the scan command on D3 it returns no devices (Max device found) if I scan D4 (the DS18B20 is still attached to D3) it finds the DS18B20 and it works just fine. However, now I can't use D4 for my input as the pin is in-use.

Swapping the devices between D pins does not help. 

This is an error in the onewire scan someplace or deep inside busio...

[code.txt](https://github.com/adafruit/circuitpython/files/5521179/code.txt)

tannewt, 2020-11-12:
@wifijt This sounds like the pin mapping is wrong. Does it work if you treat D3 as D4 and vice versa?

wifijt, 2020-11-12:
It appears that this is ONLY related to OneWire. If I treat the D3 and D4 as inputs and outputs they are mapped correctly. When I scan with onewire it is always opposite of what I am attached to: Connected on D4 - scan D3 and it finds the DS18B20 - Connect on D3 scan D4 and it finds the DS18B20 - scan the pin you are on: connected D4 scan D4 you get too many results (I've increased the max results and I get a return of a bunch of results but no DS18B20)

tannewt, 2020-11-12:
This is very weird because OneWire is bitbanged using DigitalInOut under the hood: https://github.com/adafruit/circuitpython/blob/main/shared-module/bitbangio/OneWire.c

wifijt, 2020-11-18:
This may be a board issue - just tested this on a regular sized pyportal and it works as expected.

Going to get my pynt swapped out and see if it follows on the pynt

dhalbert, 2021-03-31:
@wifijt Did you ever figure this out?

ladyada, 2021-03-31:
the silkscreen is wrong, its documented in product page

dhalbert, 2021-03-31:
> the silkscreen is wrong, its documented in product page

Aha - thanks! OK, closing.

closed 2021-03-31 by dhalbert