adafruit/circuitpython #3679: SPI Invalid Pins on Samd51 for Sparkfun's MicroMod

mrmakeit, 2020-11-11:
Trying to setup Sparkfun's MicroMod has resulted in a "Invalid Pins" error when setting up the SPI Flash.  It appears that the MicroMod uses a different pinout for the second SPI bus?  
I've checked against the Samd51 Thing Plus, and the [schematic released by Sparkfun](https://cdn.sparkfun.com/assets/0/7/2/c/4/SparkFun_MicroMod_SAMD51_Schematic.pdf).  It seems all for pins are different for the MicroMod from the Thing Plus?  Not sure where to proceed from here...

mrmakeit, 2020-11-11:
[Additional details](https://github.com/sparkfun/MicroMod_Processor_Board-SAMD51/blob/acd61f500bbd74bcf05c4084bba7b30c445b3561/Arduino%20Board%20Files/SparkFun_SAMD51_MicroMod/variant.cpp#L124) from the Sparkfun Repo to show the pins used:

tannewt, 2020-11-12:
What version of CircuitPython are you using? Are you adding the MicroMod as a board to CircuitPython? Please link me to the code you are using to test with. Thanks!

dhalbert, 2021-03-31:
Closing for now due to no activity. If you would like to add the MicroMod as a board and are getting stuck, let us know. We would welcome a PR for the board.