adafruit/circuitpython #3685: pulseio interferes with the display of the matrixportal

diegomoise1, 2020-11-12:
`import board
import displayio
import framebufferio
import rgbmatrix
import terminalio
import pulseio
displayio.release_displays()

matrix = rgbmatrix.RGBMatrix(
    width=64, bit_depth=4,
    rgb_pins=[
        board.MTX_R1,
        board.MTX_G1,
        board.MTX_B1,
        board.MTX_R2,
        board.MTX_G2,
        board.MTX_B2
    ],
    addr_pins=[
        board.MTX_ADDRA,
        board.MTX_ADDRB,
        board.MTX_ADDRC,
        board.MTX_ADDRD
    ],
    clock_pin=board.MTX_CLK,
    latch_pin=board.MTX_LAT,
    output_enable_pin=board.MTX_OE

    )
display = framebufferio.FramebufferDisplay(matrix)
#display starts blinking right here
pulses = pulseio.PulseIn(board.A1, maxlen=200, idle_state=True)
`

I am trying to get my board to receive an IR code, but as soon as a pulseio object gets declared the display starts blinking. My code is longer than what I have here but using just what I put here in the terminal I can get it to start blinking so it has nothing to do with the rest of my code. I am not sure if this is limited to the Matrixportal or if the same happens with other display or boards.

tannewt, 2020-11-12:
@jepler Does this make you think of anything suspicious?

diegomoise1, 2020-11-13:
Should I be scared that this is marked as a Long term milestone...?

tannewt, 2020-11-13:
It is not a high priority for us. If it's something obvious to someone who sees the issue, then we'll fix it. Otherwise, it'll have to wait until someone can look deeper into it.

Let us know if you'd like to help debug it.