adafruit/circuitpython #3686: OSError 0 with MagTag

makermelissa, 2020-11-12:
I was running code to fetch bitcoin and came across this mysterious error.

```
Traceback (most recent call last):
  File "code.py", line 82, in <module>
  File "code.py", line 74, in <module>
  File "/lib/adafruit_requests.py", line 585, in get
  File "/lib/adafruit_requests.py", line 563, in request
  File "/lib/adafruit_requests.py", line 423, in _get_socket
OSError: 0
```

makermelissa, 2020-11-12:
Ok cool, will do next time.

FoamyGuy, 2020-11-12:
I'm not sure if it was a fluke or not. But I also saw this error several times on the UM Feather S2 pulling from Hackster news API. Running the same code on Metro ESP32S2 I never ran into this error at all. 

Even on the Feather S2 it was not consistently this error though, many of the requests still succeeded. So it's possible I just got "luckier" on the Metro.

makermelissa, 2020-11-12:
Right, the code ran for a while before I noticed it was stopped on this error.

anecdata, 2020-11-16:
Confirming that this happens on various ESP32-S2 boards, usually after running for some hours or days. Other RuntimeErrors, like `RuntimeError: Unable to read HTTP response.`, are recoverable if caught. But this one does not seem to be, once it happens, no more internet.

jepler, 2020-11-17:
Discussing this indirectly on Discord, @tannewt said
> consider OSError(0) to mean I meant to figure out the appropriate error number

hierophect, 2021-02-10:
Closing this as it's almost certainly overridden by major reworks to the Socket module and more minor changes to Requests. Other issues are now tackling remaining instances of `OSError: 0`

closed 2021-02-10 by hierophect