adafruit/circuitpython #3693: esp32-s2 gamepad.get_pressed() stuck on 0 after restart

FoamyGuy, 2020-11-15:
Using this script:
```python
import time
import board
from digitalio import DigitalInOut, Direction, Pull
from gamepad import GamePad

print("wait before anything")
time.sleep(5)

btn_a = DigitalInOut(board.BUTTON_A)
btn_b = DigitalInOut(board.BUTTON_B)

# um feather s2 with OLED featherwing
# btn_a = DigitalInOut(board.IO33)
# btn_b = DigitalInOut(board.IO38)

_buttons = GamePad(
    btn_a,
    btn_b,
)
while True:
    print("{} - {}".format(btn_a.value, btn_b.value))
    print(_buttons.get_pressed())
    time.sleep(1)
```
When I run this on a ESP32-S2 (MagTag and UM Feather S2) directly after it boots up the printouts are as expected. The `True` and `False` values print correctly, as well as the numbers from `_buttons.get_pressed()` have the correct value.

But if I restart this script by using ctrl-c then ctrl-d the values from `_buttons.get_pressed()` will start to always show `0`. The values from `btn_a.value` and  `btn_b.value` still correctly change from True to False when I press buttons. But the gamepad values seem to get stuck on `0` after restarting this way. 

The same code run on a CLUE and prints correct values both before and after restarting with ctrl-c / ctrl-d.

jepler, 2020-11-19:
I had thought maybe the tick disable problem in #3710 could have fixed/improved this.  However, now this seems to be resetting the magtag each time I reload the test program.  Not good.  I did my testing with the artifact from #3718 which doesn't have the fix from #3716 which is causing general instability so maybe not a good test.

deshipu, 2021-02-28:
This could be somehow related to the long-lived object handling. The gampead singleton is created as a long-lived object, maybe it's not getting reset properly? I can't see anything immediately obvious in the code, though.

tannewt, 2021-07-15:
Superseded by `keypad`.

closed 2021-07-15 by tannewt