adafruit/circuitpython #3699: E-Paper Display OnDiskBitmap drawing order issue

FoamyGuy, 2020-11-16:
In this code the OnDiskBitmap gets added to the group after a Bitmap, so it should get drawn on top.
```python
import time
import board
import displayio
import adafruit_imageload

display = board.DISPLAY
time.sleep(5)

splash = displayio.Group(max_size=20)
# background

rect = displayio.Bitmap(20,20, 2)
rect_palette = displayio.Palette(3)
rect_palette[0] = 0xFFFFFF
# Create a TileGrid using the Bitmap and Palette
rect_tile_grid = displayio.TileGrid(rect, pixel_shader=rect_palette)
rect_tile_grid.x = 3
rect_tile_grid.y = 5

with open("/thermometer.bmp", "rb") as f:

    thermometer_bitmap = displayio.OnDiskBitmap(f)
    tilegrid = displayio.TileGrid(thermometer_bitmap, pixel_shader=displayio.ColorConverter())

    tilegrid.x = 10
    tilegrid.y = 10

    splash.append(rect_tile_grid)
    splash.append(tilegrid)

    display.show(splash)
    display.refresh()

    print("loop")
    while True:
        time.sleep(1)
```
But when it runs the Bitmap seems to get drawn on top:
[image]

Using very similar code with adafruit_imageload instead of OnDiskBitmap:
```python
import time
import board
import displayio
import adafruit_imageload

display = board.DISPLAY
time.sleep(5)

splash = displayio.Group(max_size=20)
# background

rect = displayio.Bitmap(20,20, 2)
rect_palette = displayio.Palette(3)
rect_palette[0] = 0xFFFFFF
# Create a TileGrid using the Bitmap and Palette
rect_tile_grid = displayio.TileGrid(rect, pixel_shader=rect_palette)
rect_tile_grid.x = 3
rect_tile_grid.y = 5

image, palette = adafruit_imageload.load(
    "/thermometer.bmp", bitmap=displayio.Bitmap, palette=displayio.Palette
)
tilegrid = displayio.TileGrid(image, pixel_shader=palette)

tilegrid.x = 10
tilegrid.y = 10

splash.append(rect_tile_grid)
splash.append(tilegrid)

display.show(splash)
display.refresh()

print("loop")
while True:
    time.sleep(1)
```
The image does get drawn on top:
[image]
The thermometer image being used is here: 
[thermometer.bmp.zip](https://github.com/adafruit/circuitpython/files/5543647/thermometer.bmp.zip)

ladyada, 2020-12-22:
@FoamyGuy we recently fixed a few things with displayio and eink, do you want to try again?

FoamyGuy, 2020-12-22:
Yep, I will try this out tonight.

FoamyGuy, 2020-12-23:
Indeed this is fixed using the newest from S3. I know it was still an issue on the beta.2 release, but I didn't try the ones in between, not sure exactly where the fix came.
```
Adafruit CircuitPython 6.1.0-beta.2-171-g8b03951fe on 2020-12-22; Adafruit MagTag with ESP32S2
```

closed 2020-12-23 by FoamyGuy