adafruit/circuitpython #3700: Booting from the SD card

kahveciderin, 2020-11-16:
Title says it all, I don't want the boot.py/main.py on the internal flash, rather in the SD card. How may I accomplish this?

tannewt, 2020-11-16:
We don't support it because the SD card may not exist. You can use `import` to load code off the SD card from `main.py`. https://github.com/adafruit/circuitpython/pull/3454 also provides a new way to reload the VM and run a different file.

closed 2020-11-16 by tannewt

kahveciderin, 2020-11-19:
@tannewt  soo can I put the precompiled python data to the internal flash on the samd51 and tell the samd to execute a file from the sd card?

ladyada, 2020-11-19:
there isnt 'precompiled python' in the sense that it is executed. python is interpretted only. you can use mpy-cross to mpy'ify a py file, that only makes it more compact (bytecode conversion). then import it as any other python code

tannewt, 2020-11-19:
@kahveciderin I'm not sure what you are trying to accomplish.

kahveciderin, 2020-11-19:
@ladyada yeah that was what i meant

@tannewt i am trying to execute a file from the sd card, without the external flash.

tannewt, 2020-11-23:
@kahveciderin Please explain the larger context. I want to understand why you are trying to execute without internal flash.

We don't have a way to execute from the external flash without something on the built in flash fs.

kahveciderin, 2020-11-26:
@tannewt let's say I have soldered the flash chip out. I want to execute from the sd card. Is this possible?

deshipu, 2020-11-26:
You can compile a version that uses a part of the internal flash for the filesystem.

tannewt, 2020-12-01:
@kahveciderin It is not possible with the current code. As deshipu points out, you could have a code.py on an internal filesystem to configure and run off the sd card.

kahveciderin, 2021-02-10:
@tannewt do I need to edit the ```run_code_py``` function in the main.c file?

tannewt, 2021-02-10:
@kahveciderin It depends on how you are mounting the SD card. If you mount it to / then you shouldn't need to.