adafruit/circuitpython #3701: Broken S3 update for feather_nrf52840_express

smurfix, 2020-11-16:
The "lastest" packages here are two years old:

https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/feather_nrf52840_express/

I need 6.1.0.alpha for grayscale eInk support; building it myself unfortunately didn't work (stayed in FTHR840BOOT after flashing) and I have no idea how to figure out what went wrong / didn't yet buy a JTAG adapter.

FoamyGuy, 2020-11-16:
I think you need to click into one of the languages near the top. Here is the link for en_US: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/feather_nrf52840_express/en_US/

Those contains the newer builds because they are now translated into many languages.

smurfix, 2020-11-16:
*Sigh* I should have noticed. Thx.

closed 2020-11-16 by smurfix