adafruit/circuitpython #3709: Can't remount in boot.py anymore

Neradoc, 2020-11-17:
So I'm encountering the following with the latest build from S3
The following boot.py (no matter if using True or False)

```import storage
storage.remount("/", False)```

Results in the following boot_out.txt

```Adafruit CircuitPython 6.1.0-beta.0 on 2020-11-17; FeatherS2 with ESP32S2
boot.py output:
Traceback (most recent call last):
  File "boot.py", line 2, in <module>
RuntimeError: Cannot remount '/' when USB is active.```

kattni, 2020-12-21:
Adding info from #3833 - storage.remount("/", True) resulting in RuntimeError on 6.1.0-beta.2 

When working through [this CircuitPython storage example](https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage) on QT Py M0, running 6.1.0-beta.2, it fails to work properly - the temperature logging file is never created and the filesystem never switches to read-only. Checking `boot_out.txt`, I find the following traceback:
```
Traceback (most recent call last):
  File "boot.py", line 18, in <module>
RuntimeError: Cannot remount '/' when USB is active.
```
I tried attaching it to a USB battery pack, thinking it meant the USB data connection - this did not resolve the issue, the results are the same. Tried the example on a Trinket running 6.1.0-beta.2, and the results are the same.

I tried the QT Py and Trinket running 6.0.0, and the example works properly.

closed 2020-12-22 by tannewt

vivianasalcedo, 2021-12-29:
The solution they presented to store the microcontroller temperature in a file does not work. Although it was written in the boot.py storage.remount("/", FALSE), it does not work in a cyclic way, it does it only once, that is, it only saves the temperature once. I wonder how to solve this problem?
The solution they presented to store the microcontroller temperature in a file does not work. Although it was written in the boot.py storage.remount("/", FALSE), it does not work in a cyclic way, it does it only once, that is, it only saves the temperature once. I wonder how to solve this problem?

dhalbert, 2021-12-29:
We think we fixed the bug. If you think there is a new bug, could you file an issue with complete details, or could you ask for help in discord: https://adafru.it/discord or the forums: https://forums.adafruit.com/viewforum.php?f=60 ? We need to see the code you're using and the versions you're using.

vivianasalcedo, 2021-12-29:
I don't think it is corrected, in the boot.py is the following code "short version":
storage.remount("/", FALSE)

It is assumed that when it is FALSE in the above command, then a file can be opened and written to. 

while TRUE:
             file = open("temperature.txt", "wr")
             file.write(str(int(int(microcontroller.cpu.temperature))))
             file.close()

dhalbert, 2021-12-29:
Are you using `FALSE`/`TRUE` or `False`/`True`. The latter is correct.

Please show your whole code, both boot.py and code.py.

Neradoc, 2021-12-29:
> Although it was written in the boot.py storage.remount("/", FALSE), it does not work in a cyclic way, it does it only once, that is, it only saves the temperature once. I wonder how to solve this problem?

Booleans notwithstanding, that is to be expected, `open(...,"wr")` will truncate the file, so you overwrite it with the last temperature every time. You want `"a"` for append.

vivianasalcedo, 2021-12-29:
boot.py : 
import board
import digitalio
import storage

cambio =  digitalio.DigitalInOut(board.D2)
cambio.direction = digitalio.Direction.INPUT
cambio.pull = digitalio.Pull.UP
storage.remount("/", cambio.value)
print(cambio.value)

vivianasalcedo, 2021-12-29:
code.py:
import microcontroller
import time
import board
import digitalio

while True:
    temp = microcontroller.cpu.temperature
    file = open("temperatura.txt","wr")
    file.write('{0:f}\n'.format(int(temp)))
    file.close()
    time.sleep(3)

dhalbert, 2021-12-29:
As @Neradoc mentions, change `"wr"` to `"wa"`:
```py
    file = open("temperatura.txt","wr")
```

vivianasalcedo, 2021-12-29:
not working

dhalbert, 2021-12-29:
I see you on discord. Let's discuss there rather than here.

vivianasalcedo, 2021-12-29:
which means wa

vivianasalcedo, 2021-12-29:
unfortunately it doesn't work, do you have more examples of how to save files in circuitpython?

vivianasalcedo, 2021-12-30:
Thanks for your help Neradoc and dhalbert, I did it again with more calm and I really wrote again in the same line, now I can add the data to the .txt file.

PR #3862 (2020-12-22): unmounted filesystems start as ejected
Fixes #3709.

Undoes https://github.com/adafruit/circuitpython/commit/bda3267432aee57e704a20308b724369cceeb6ca from #3680. The problem is that a board that starts unconnected does not start with unejected drives; they should be ejected when `boot.py` is entered.

The original change saved 20 bytes. I tried reversing the sense of the `ejected[]` array, calling it `not_ejected[]`.  (`mounted` was too confusing, because it has two meanings: one for TinyUSB, and one for the filesystem.) But the code was harder to read and the change caused the addition of a l [... cut]