adafruit/circuitpython #3712: ESP32S2 WiFi connect() hard-resets and "Connection Error: Unknown failure"

todbot, 2020-11-18:
Tested on Metro ESP32-S2 and MagTag. This was also present in a debug firmware for MagTag provided by @tannewt on 17 Nov 2020. 

On CircuitPython-6.1.0-beta.0, the `wifi` module will sometimes hard-reset on `wifi.radio.connect()`, using a small connect script (*)  This seems to only occur on Ctrl-D and save-on-reload restarts. When starting from hardware reset / power-on, the script works.

Also on CP-6.1.0-beta.0, it seems the ESP32-S2 WiFi cannot connect to APs that are not DHCP servers (but instead use a DHCP server over LAN).  I have two APs on my network: one is the router and DHCP server, another is secondary AP (with different SSID) on the same LAN as the DHCP server.  The ESP32-S2 CP can connect to the primary AP that's the DHCP server, but will error out with `Connection Error: Unknown failure` on the secondary AP.  This almost feels like the ESP32-S2 is timing out early for a DHCP response.

*e.g. a script such as:
```py
import wifi
for network in wifi.radio.start_scanning_networks():
    print("\t'%s'\t\tRSSI: %d\tChannel: %d" % (network.ssid, network.rssi, network.channel))
wifi.radio.stop_scanning_networks()
print("connecting...")
wifi.radio.connect(ssid="mysssid", password="supersecret")
print("connected.")
```

todbot, 2020-11-18:
After doing tests with Metro ESP32-S2 with the Arduino IDE, the second part of this issue (`Connection Error: Unknown failure` on a secondary AP) is not unique to CircuitPython.  Perhaps ESP32-S2s cannot connect to Apple Airports in AP mode or to APs that forward DHCP over Ethernet.

todbot, 2020-11-18:
Small update: upon reboot of problematic AP (Apple Time Capsule w/ external DHCP), Metro ESP32-S2 connection results:
- Arduino ESP32-S2 "idf-release/v4.2" : connects
- CircuitPython 20201011-40a3d11: connects
- CircuitPython 6.1.0-beta.0: **does not** connect, and often hard-resets

tannewt, 2020-11-19:
beta.0 is broken. Please use the a18 build until #3716 is merged into main.

anecdata, 2020-11-25:
Helping with a connection issue on Discord ([https://discord.com/channels/327254708534116352/537365702651150357/781234430520852491](https://discord.com/channels/327254708534116352/537365702651150357/781234430520852491)), I was able to connect to an Apple Airport Express set u as "Create a Network" in bridge mode, with 
```
FeatherS2 with ESP32S2
6.1.0-beta.1 on 2020-11-20
adafruit-circuitpython-bundle-6.x-mpy-20201120
```

tannewt, 2020-11-25:
@todbot Any update? Maybe we can close this.

todbot, 2020-11-26:
Hi @tannewt, I've not been able to reproduce this error on `6.1.0-beta.1` on Metro ESP32S2.  So okay for this to be closed I guess. Thanks!

closed 2020-11-26 by todbot