adafruit/circuitpython #3715: esp32s2: spi (and other peripherals?) don't claim pins properly

jepler, 2020-11-18:
I noticed that it was possible to create a DigitalIO from some pin while the pin was already in use by an SPI instance:
```
>>> spi = busio.SPI(board.IO0, board.IO1, board.IO2)
>>> import digitalio
>>> pin = digitalio.DigitalInOut(board.IO0)
>>> pin
<DigitalInOut>
```

tested on a kaluga with a 6.1-ish firmware

microDev1, 2020-11-19:
I can confirm this bug to be present on other peripherals and not just related to spi.

hierophect, 2020-11-24:
Almost certainly related to #3643

hierophect, 2020-11-24:
Actually, looks like I misread - the reset problem will cause conflicts pins that have been released, but it shouldn't allow pins in the claimed array to be claimed *again*. This could be an issue with the bit shifting in the pin claiming system.

hierophect, 2020-11-24:
@microDev1 can you point out specific modules where you have also had this problem? Turns out this was just that the SPI module didn't have pin claiming in it. The other Busio modules have it, as does DigitalIO and AnalogIO, but if you've encountered it elsewhere I can fold a fix into my PR.

microDev1, 2020-11-25:
@hierophect I am able to see this on basically all of the io modules.

[image]

microDev1, 2020-11-25:
Update... Pin claiming seems to work until `IO7` after which pins aren't claimed properly.

hierophect, 2020-11-25:
@microDev1 found it, it was a variable size issue with in pin claiming mask operations.

PR #3754 (2020-11-25): ESP32S2: fix pin claiming bugs
Resolves #3715. Tested on a Saola Wrover.

Edit: now fixes two bugs, the first being that SPI was missing the `claim_pin` functions, and the second being that a mask variable being too small in microcontroller/Pin was preventing IO pins over 7 from being reserved.
closed 2020-11-25 by tannewt