adafruit/circuitpython #3717: Crash with LoadInstProhibited

tannewt, 2020-11-19:
I attached to the native UART on MagTag. These are the backtraces I got when the usb disappeared or the serial connection hung.

```
? Backtrace:0xfffffffd:0x3ffe9280 0x400e2bc9:0x3ffe92a0 0x40036827:0x3ffe92c0 0x40037c4d:0x3ffe93e0 0x4003b6f5:0x3ffe9400 0x4003982b:0x3ffe9440
got ['0xfffffffd', '0x400e2bc9', '0x40036827', '0x40037c4d', '0x4003b6f5', '0x4003982b']
??:0
/home/tannewt/repos/circuitpython/ports/esp32s2/build-adafruit_magtag_2.9_grayscale/esp-idf/../../esp-idf/components/esp_wifi/src/wifi_netif.c:47
??:?
??:?
??:?
??:?
? Backtrace:0xfffffffd:0x3ffe8b00 0x400e2c79:0x3ffe8b20 0x40036827:0x3ffe8b40 0x40037c4d:0x3ffe8c60 0x4003b6f5:0x3ffe8c80 0x4003982b:0x3ffe8cc0
got ['0xfffffffd', '0x400e2c79', '0x40036827', '0x40037c4d', '0x4003b6f5', '0x4003982b']
??:0
/home/tannewt/repos/circuitpython/ports/esp32s2/build-adafruit_magtag_2.9_grayscale/esp-idf/../../esp-idf/components/esp_wifi/src/wifi_netif.c:50 (discriminator 15)
??:?
??:?
??:?
??:?
```

tannewt, 2020-11-24:
UART output:

```
I (4804459) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (4804469) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
W (4804539) cp radio: connect()
I (4805269) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (4805739) wifi:state: init -> auth (b0)
I (4805739) wifi:state: auth -> assoc (0)
I (4806749) wifi:state: assoc -> init (400)
I (4806749) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
W (4806749) cp wifi: disconnected
W (4806749) cp wifi: reason 4 0x04
I (4806749) cp wifi: Retrying connect. 4 retries remaining
I (4808089) wifi:new:<11,0>, old:<6,0>, ap:<255,255>, sta:<11,0>, prof:1
I (4808089) wifi:state: init -> auth (b0)
I (4808099) wifi:state: auth -> assoc (0)
I (4808099) wifi:state: assoc -> run (10)
I (4808109) wifi:connected with skylab, aid = 3, channel 11, BW20, bssid = 74:83:c2:73:49:ee
I (4808109) wifi:security: WPA2-PSK, phy: bgn, rssi: -67
I (4808109) wifi:pm start, type: 1

I (4808119) wifi:AP's beacon interval = 102400 us, DTIM period = 1
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x800e3950  A1      : 0x3ffe8c40
A2      : 0x3ffff048  A3      : 0x3ffdf3b2  A4      : 0x00000082  A5      : 0x3ffdf318
A6      : 0x0000e949  A7      : 0x00000000  A8      : 0x80158f74  A9      : 0x0000008c
A10     : 0x00000000  A11     : 0x3ffdf3b2  A12     : 0x00000082  A13     : 0x3ffdf318
A14     : 0x3ffe79c0  A15     : 0xffffffff  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x00000082  LEND    : 0x3ffdf318  LCOUNT  : 0x40029371

Backtrace:0xfffffffd:0x3ffe8c40 0x400e394d:0x3ffe8c60 0x4003693b:0x3ffe8c80 0x40037d61:0x3ffe8da0 0x4003b809:0x3ffe8dc0 0x4003993f:0x3ffe8e00
```

Similar backtrace as before:
```
??:0
/home/tannewt/repos/circuitpython/ports/esp32s2/build-adafruit_magtag_2.9_grayscale/esp-idf/../../esp-idf/components/esp_wifi/src/wifi_netif.c:47
??:?
??:?
??:?
??:?
```

BennyE, 2020-12-12:
The step after "I (4808119) wifi:AP's beacon interval = 102400 us, DTIM period = 1" is to obtain an IP address and initialisation of the TCP/IP stack. I believe you'll not see this issue again, as it was related to the duplicated esp_netif_init() and or the associated esp_event_loop_create_default() which @jepler fixed via https://github.com/adafruit/circuitpython/commit/dd108b755dc2bb886df14c9fcfcd4f09be61f2d4#diff-d3fbda07013b552de603fbc99e263532c77942ec1f4746a4b5aa755f7d3ae640

tannewt, 2020-12-15:
Ok, thanks @BennyE. I'll close. We can always reopen if we see it again.

closed 2020-12-15 by tannewt