adafruit/circuitpython #3726: Hard reset on second use of wifi (MagTag / ESP32S2)

caternuson, 2020-11-19:
This is a modified version of the bitcoin example:
```python
from adafruit_magtag.magtag import MagTag

DATA_SOURCE = "https://api.coindesk.com/v1/bpi/currentprice.json"
DATA_LOCATION = ["bpi", "USD", "rate_float"]

magtag = MagTag(
    url=DATA_SOURCE,
    json_path=DATA_LOCATION,
)

magtag.network.connect()

resp = magtag.fetch()
print("Bitcoin =", resp)
```
which is saved as `fetch_test.py` to the MagTag's `CIRCUITPY` folder. Then, connect to REPL and:
```
Adafruit CircuitPython 6.1.0-beta.0-13-g3e6661bc9 on 2020-11-19; MagTag with ESP32S2
>>> import fetch_test
Connecting to AP ----
Retrieving data...Reply is OK!
Bitcoin = 18023.4
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.1.0-beta.0-13-g3e6661bc9 on 2020-11-19; MagTag with ESP32S2
>>> import fetch_test
Connecting to AP ----
```
**hard resets**

caternuson, 2020-11-19:
Sounds like this might be related to something known.

Issue does **not** occur with `adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-20201117-a185549.bin`

tannewt, 2020-11-19:
This should be fixed by #3716

caternuson, 2020-11-20:
Indeed it does. Thanks!

```
Adafruit CircuitPython 6.1.0-beta.0-25-g2f1460904 on 2020-11-19; MagTag with ESP32S2
>>> import fetch_test
Connecting to AP ----
Retrieving data...Reply is OK!
Bitcoin = 17899.7
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Adafruit CircuitPython 6.1.0-beta.0-25-g2f1460904 on 2020-11-19; MagTag with ESP32S2
>>> import fetch_test
Connecting to AP ----
Retrieving data...Reply is OK!
Bitcoin = 17916.9
>>> 
```

closed 2020-11-20 by caternuson