adafruit/circuitpython #3731: start_advertising throws _bleio.BluetoothError: Unknown soft device error: 0008 for certain conditions in 6.0.0

kevinjwalters, 2020-11-20:
I just fired up [Adafruit Learn: CLUE Rock, Paper, Scissors Game using Bluetooth](https://learn.adafruit.com/rock-paper-scissors-circuitpython) in 6.0.0 with a fresh (20201120) set of 6.x libraries and it throws an exception as soon as it starts the relatively complicated bluetooth chatter on a CLUE.

I tried a bit of REPLing to reproduce a small example of it. This seems ok:

```
Adafruit CircuitPython 6.0.0 on 2020-11-16; Adafruit CLUE nRF52840 Express with nRF52840
>>> from adafruit_ble import BLERadio
>>> from adafruit_ble.advertising.adafruit import AdafruitColor
>>> ca = AdafruitColor()
>>> ca.color = 0x445566
>>> radio = BLERadio()
>>> radio.start_advertising(ca)
>>> radio.stop_advertising()
>>> radio.start_advertising(ca)
>>> radio.stop_advertising()
>>> radio.start_advertising(ca)
>>> radio.stop_advertising() ; radio.start_advertising(ca)
```

But this is not ok:

```
Adafruit CircuitPython 6.0.0 on 2020-11-16; Adafruit CLUE nRF52840 Express with nRF52840
>>> from adafruit_ble import BLERadio
>>> from adafruit_ble.advertising.adafruit import AdafruitColor
>>> ca = AdafruitColor()
>>> ca.color = 0x445566
>>> radio = BLERadio()
>>> radio.start_advertising(ca)
>>> for adv in radio.start_scan(timeout=30.0):
...     print(adv)
...     radio.stop_advertising()
...     radio.start_advertising(ca)
...
...
...
<Advertisement flags=<AdvertisingFlags general_discovery > >
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "adafruit_ble/__init__.py", line 212, in start_advertising
_bleio.BluetoothError: Unknown soft device error: 0008
>>>
```

I'm relying on general BLE traffic in my vicinity to cause the loop to iterate and as can be seen from output first attempt at stop/start blows up. I've tried this 3 times from application and 4 times on REPL and the exception is predictable so far.

tannewt, 2020-12-29:
@dhalbert Do you think this was fixed in 6.0.1 as well?

dhalbert, 2020-12-30:
> @dhalbert Do you think this was fixed in 6.0.1 as well?

I don't know of a specific fix, but @kevinjwalters, could you try it with 6.0.1? Thanks.

kevinjwalters, 2021-01-02:
Bug is still present:

```
Adafruit CircuitPython 6.0.1 on 2020-12-28; Adafruit CLUE nRF52840 Express with nRF52840
>>> from adafruit_ble import BLERadio
>>> from adafruit_ble.advertising.adafruit import AdafruitColor
>>> ca = AdafruitColor()
>>> ca.color = 0x4b4c46
>>> radio = BLERadio()
>>> radio.start_advertising(ca)
>>> for adv in radio.start_scan(timeout=30.0):
...     print(adv)
...     radio.stop_advertising()
...     radio.start_advertising(ca)
...
...
...
<Advertisement tx_power=12 flags=<AdvertisingFlags general_discovery > >
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "adafruit_ble/__init__.py", line 212, in start_advertising
_bleio.BluetoothError: Unknown soft device error: 0008
```

dhalbert, 2021-01-04:
This error is:
```c
#define NRF_ERROR_INVALID_STATE               (NRF_ERROR_BASE_NUM + 8)  ///< Invalid state, operation disallowed in this state
```

tannewt, 2021-01-21:
@dhalbert did this get fixed?

dhalbert, 2021-01-21:
No, I haven't looked at it yet.

kevinjwalters, 2022-04-16:
`7.2.0` doesn't like a `start_advertising()` inside a for loop reading from a `start_scan()`:

```
Traceback (most recent call last):
  File "code.py", line 297, in <module>
  File "adafruit_ble/__init__.py", line 195, in start_advertising
_bleio.BluetoothError: Unknown system firmware error: 0008
```