adafruit/circuitpython #3735: Pool, Request, IO_HTTP reusabilit

askpatrickw, 2020-11-21:
Attached is a very slimmed down example. I was doing more sending data from a sensor, but you can reproduce this just using the time server on AIO.
[sleep_bug.txt](https://github.com/adafruit/circuitpython/files/5576938/sleep_bug.txt)

```python
    ## SETUP BLOCK ##
    print("Setup Requests & IO_HTTP")
    POOL = socketpool.SocketPool(wifi.radio)
    REQUESTS = adafruit_requests.Session(POOL, ssl.create_default_context())
    # Initialize an Adafruit IO HTTP API object
    AIO = IO_HTTP(config.ADAFRUIT_IO_USERNAME, config.ADAFRUIT_IO_KEY, REQUESTS)
    ## SETUP BLOCK ##
```
### POOL, REQUESTS AND AIO inside the While works 1 time.

```python
code.py output:
Get on the network
ip: 192.168.1.246
hostname: Metro-ESP32S2
Setup Requests & IO_HTTP

Ping: 0.016
AIO Time: struct_time(tm_year=2020, tm_mon=11, tm_mday=20, tm_hour=19, tm_min=25, tm_sec=21, tm_wday=5, tm_yday=325, tm_isdst=0)
Sleeping for 30 seconds.

Ping: 0.013
Traceback (most recent call last):
  File "code.py", line 31, in <module>
  File "/lib/adafruit_io/adafruit_io.py", line 701, in receive_time
  File "/lib/adafruit_io/adafruit_io.py", line 542, in _get
  File "adafruit_requests.py", line 594, in get
  File "adafruit_requests.py", line 581, in request
  File "adafruit_requests.py", line 119, in __init__
RuntimeError: Unable to read HTTP response.
```

Debug Output from Kaluga

```
W (9669) wifi: got ip
I (10639) esp-x509-crt-bundle: Certificate validated
I (102699) wifi:state: run -> init (0)
I (102699) wifi:pm stop, total sleep time: 92188662 us / 95776179 us

I (102699) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
W (102699) wifi:hmac tx: stop, discard
W (102709) wifi: disconnected
W (102709) wifi: reason 8 0x08
I (102739) wifi:flush txq
I (102739) wifi:stop sw txq
I (102739) wifi:lmac stop hw txq
I (102739) wifi:Deinit lldesc rx mblock:4
I (102749) gpio: GPIO[45]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```

### POOL, REQUESTS AND AIO inside the While works 2 times.

```python
code.py output:
Get on the network
ip: 192.168.1.246
hostname: Metro-ESP32S2

Ping: 0.011
Setup Requests
AIO Time: struct_time(tm_year=2020, tm_mon=11, tm_mday=20, tm_hour=19, tm_min=22, tm_sec=46, tm_wday=5, tm_yday=325, tm_isdst=0)
Sleeping for 30 seconds.

Ping: 0.013
Setup Requests
AIO Time: struct_time(tm_year=2020, tm_mon=11, tm_mday=20, tm_hour=19, tm_min=23, tm_sec=19, tm_wday=5, tm_yday=325, tm_isdst=0)
Sleeping for 30 seconds.

Ping: None
Setup Requests
Traceback (most recent call last):
  File "code.py", line 28, in <module>
  File "/lib/adafruit_io/adafruit_io.py", line 701, in receive_time
  File "/lib/adafruit_io/adafruit_io.py", line 542, in _get
  File "adafruit_requests.py", line 594, in get
  File "adafruit_requests.py", line 572, in request
  File "adafruit_requests.py", line 441, in _get_socket
RuntimeError: Sending request failed
```

Debug output from the Kaluga

```
W (248669) wifi: got ip
I (249639) esp-x509-crt-bundle: Certificate validated
I (282819) esp-x509-crt-bundle: Certificate validated
E (318099) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7f00
E (318099) esp-tls: create_ssl_handle failed
E (318099) esp-tls: Failed to open new connection
I (318179) wifi:state: run -> init (0)
I (318179) wifi:pm stop, total sleep time: 67591551 us / 72243520 us

I (318179) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
W (318179) wifi:hmac tx: stop, discard
W (318189) wifi:hmac tx: stop, discard
W (318189) wifi: disconnected
W (318189) wifi: reason 8 0x08
I (318229) wifi:flush txq
I (318229) wifi:stop sw txq
I (318229) wifi:lmac stop hw txq
I (318229) wifi:Deinit lldesc rx mblock:4
I (318239) gpio: GPIO[45]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```

tannewt, 2020-11-24:
I wonder if this is a failure of the socketpool. I don't think we handle reconnects very well currently.

askpatrickw, 2020-12-21:
If I move the network setup outside the loop... this runs indefinitely.
I wonder if re-re-initializing the network is creating a mess.... 

```python
....
print("Get on the network")
wifi.radio.connect(config.SSID, config.WIFI_PASSWORD)
print(f"ip: {wifi.radio.ipv4_address}")
print(f"hostname: {wifi.radio.hostname}")

## SETUP BLOCK ##
print("Setup Requests & IO_HTTP")
POOL = socketpool.SocketPool(wifi.radio)
REQUESTS = adafruit_requests.Session(POOL, ssl.create_default_context())
# Initialize an Adafruit IO HTTP API object
AIO = IO_HTTP(config.ADAFRUIT_IO_USERNAME, config.ADAFRUIT_IO_KEY, REQUESTS)
## SETUP BLOCK ##

while True:
    print()
    print(f"Ping: {wifi.radio.ping(ipaddress.ip_address('8.8.8.8'))}")
    
    

    print(f"AIO Time: {AIO.receive_time()}")

    print(f"Sleeping for {SLEEP_TIME} seconds.")
    sleep(SLEEP_TIME)
    ```

tannewt, 2020-12-22:
What is the original bad code? Maybe sockets are being leaked when recreating the Session.

askpatrickw, 2020-12-22:
What I'm reproducing now is a wifi.radio.connect issue and possibly the original issue is fixed. 

Consistent failure on the 2nd loop fails.

```python
>>> import wifi
>>> from time import sleep
>>> import config
>>> import ipaddress
>>> while True:
...     wifi.radio.connect(config.SSID, config.WIFI_PASSWORD)
...     print(f"Ping: {wifi.radio.ping(ipaddress.ip_address('8.8.8.8'))}")
...     sleep(60)
...
```

Clean build on main
idf.py --version
ESP-IDF v4.3-dev-1198-gd06744f5e

where idf.py
.../code/github/adafruit/circuitpython/ports/esp32s2/esp-idf/tools/idf.py

askpatrickw, 2020-12-29:
Connect inside the While. None on the second ping

```
Adafruit CircuitPython 6.1.0-beta.2-151-g737828ccb-dirty on 2020-12-21; FeatherS2 PreRelease with ESP32S2
>>> import wifi
>>> from time import sleep
>>> import config
>>> import ipaddress
>>> while True:
...     wifi.radio.connect(config.SSID, config.WIFI_PASSWORD)
...     print(f"Pring: {wifi.radio.ping(ipaddress.ip_address('8.8.8.8'))}")
...     sleep(60)
...
Pring: 0.012
Pring: None
```

Connect outside the While works. This should be something we doc what goes inside\outside, what is preserved as you time.sleep (not to mention deep sleep) and then I think wifi.radio.connect should handle this more gracefully as well. 

```
Adafruit CircuitPython 6.1.0-beta.2-151-g737828ccb-dirty on 2020-12-21; FeatherS2 PreRelease with ESP32S2
>>> import wifi
>>> import config
>>> import ipaddress
>>> from time import sleep
>>> wifi.radio.connect(config.SSID, config.WIFI_PASSWORD)
>>> while True:
...     print(f"Ping: {wifi.radio.ping(ipaddress.ip_address('8.8.8.8'))}")
...     sleep(60)
...
Ping: 0.019
Ping: 0.009
Ping: 0.014
Ping: 0.01
```

tannewt, 2020-12-30:
time.sleep should leave anything running alone. So, wifi should stay up. Same for light sleep. Only deep sleep shuts things down.

I agree we should make calling connect a second time work.

askpatrickw, 2021-01-02:
I retitled this issue as new testing shows all the other cases are fixed, but calling connect more than once causes bad things to happen.

anecdata, 2021-01-17:
There's a comment at the top of common-hal `connect()`: "// check enabled"

If I'm understanding right, wifi start-stop is exclusively managed by `radio_set_enabled()` (called once per reload by wifi-Init, then can be called again by circuitpython user code to start/stop wifi). So implementing "check enabled" would not let us connect if wifi isn't started up. I suspect this is where the issue currently lies.

The other factor potentially is whether there is already a station connection to an AP. The Espressif docs don't list an `ESP_ERR` for trying to connect when already connected, so I suspect that would just be a NOP, but I can try to test that. I haven't looked into the `WIFI_CONNECTED_BIT | WIFI_DISCONNECTED_BIT`, but that's presumably a way we could check for an already=present connection in any case.

anecdata, 2021-01-17:
The early calls in `connect()` seem to be robust to getting called twice:
`esp_wifi_set_mode()  // in start_station()` & `esp_wifi_set_config()`

But calling `esp_wifi_connect()` twice in a row in `radio.c` does trigger:
```
ESP_ERR_WIFI_CONN: WiFi internal error, station or soft-AP control block wrong
```
Oddly, in CircuitPython this seems to be fine:
```
wifi.radio.connect(secrets["ssid"], secrets["password"])
wifi.radio.connect(secrets["ssid"], secrets["password"])
```
But interspersing connect with pings or GETs within the while loop does trigger `ESP_ERR_WIFI_CONN`.

I'll try some state checking early in connect to return if already connected.

anecdata, 2021-01-19:
@askpatrickw Would you mind taking a look at the PR, testing if you can. Artifacts are building now.

askpatrickw, 2021-01-19:
> @askpatrickw Would you mind taking a look at the PR, testing if you can. Artifacts are building now.

@anecdata  I can look at this 20 Jan prolly later in the day. Thanks for the diligence you took in looking into the fix!

askpatrickw, 2021-01-20:
@anecdata  Here are some things I've tested tonight, letting then run for ~ 1 minute each.

Passed
1. wifi.connect and ping in the While Loop. No issues. sleep of 5 seconds.
2. wifi.connect outside the while loop. Ping inside the while loop. Sleep of 5 seconds
3. wifi.conect, pool, and requests setup outside the while. Aio setup, aio.receive_time and ping inside the loop.
4. wifi.conect, pool, and requests AIO setup outside the while. Aio.receive_time and ping inside the loop.

The following code fails consistently.

```python
while True:
    print("Get on the network")
    wifi.radio.connect(config.SSID, config.WIFI_PASSWORD)
    print("Setup Pool")
    POOL = socketpool.SocketPool(wifi.radio)
    print("Setup Requests")
    REQUESTS = adafruit_requests.Session(POOL, ssl.create_default_context())
    print("Setup AIO")
    AIO = IO_HTTP(config.ADAFRUIT_IO_USERNAME, config.ADAFRUIT_IO_KEY, REQUESTS)

    print(f"AIO Time: {AIO.receive_time()}")

    print(f"ip: {wifi.radio.ipv4_address}")
    print(f"Ping: {wifi.radio.ping(ipaddress.ip_address('8.8.8.8'))}")
    print()    
    
    print(f"Sleeping for {SLEEP_TIME} seconds.")
    sleep(SLEEP_TIME)
```

100% Fail on the 3rd loop when setting up AIO
```
Sleeping for 5 seconds.
Get on the network
Setup Pool
Setup Requests
Setup AIO
Traceback (most recent call last):
  File "code.py", line 27, in <module>
  File "/lib/adafruit_io/adafruit_io.py", line 701, in receive_time
  File "/lib/adafruit_io/adafruit_io.py", line 542, in _get
  File "adafruit_requests.py", line 612, in get
  File "adafruit_requests.py", line 576, in request
  File "adafruit_requests.py", line 445, in _get_socket
RuntimeError: Sending request failed
```

Environment
```
circup freeze
Found device at /Volumes/CIRCUITPY, running CircuitPython 6.1.0-rc.1-18-g672d0e27a.
adafruit_ntp==0.0.0-auto.0
adafruit_ahtx0==1.0.2
adafruit_requests==1.9.2
adafruit_bme680==3.3.0
adafruit_lc709203f==2.0.0
adafruit_bus_device==5.0.3
adafruit_io==5.0.2
```

If you want me to treat that as a new issue let me know. It is getting better though. Thank you !

anecdata, 2021-01-20:
Thanks very much for stress testing! The PR mainly addresses multiple sequential connects, whether wifi is started or stopped, and a little resilience with connecting and disconnecting repeatedly.

I do think the last case is separate, since socketpool init and requests init are inside the loop (I don't know about the AIO init, but it looks like that held up in yout 3rd test)), brentru or tannewt could perhaps comment better on that... I've always kept the setups outside the loop.

askpatrickw, 2021-01-20:
I'm also fine with being told "don't do that". I do think as we get into power saving sleep it might be needed and raise the same issues, but I've not tried it yet. That is next on my todo list though.
I'll open a new issue for it and offer my approval based on the testing on the PR.

tannewt, 2021-01-20:
> I'm also fine with being told "don't do that".

We should have CircuitPython tell you not to do that. We should track if the radio already has a pool and raise an exception on the second init. We'll need to add a `deinit` to SocketPool too just in case you want to deinit and recreate. This matches how most of our APIs hold exclusive access to things.

askpatrickw, 2021-01-21:
I opened that issue separately over here in requests. It had nothing to do with AIO.
https://github.com/adafruit/Adafruit_CircuitPython_Requests/issues/62

When Dan chooses to merge this can close

closed 2021-01-22 by dhalbert