adafruit/circuitpython #3736: Feather STM32F405 Does not properly handle I2C repeated start

siddacious, 2020-11-21:
I'm working on a library and happened to grab a Feather STM32F405 to work on  and eventually found that the bug I was experiencing was due to the following code not being executed as a repeated start;

```
        buffer = bytearray([0x80, command_id])
        with self.i2c_device as i2c:
            i2c.write_then_readinto(buffer, buffer, in_end=1)
        return buffer[0]
```

Here's the trace of the STM32, flashed to CP6.1.0beta 1
[image]

And the same code run on a nRF52840 with the same CP version:
[image]

siddacious, 2020-11-21:
It looks like the flag in `common_hal_busio_i2c_write` is being ignored:
https://github.com/adafruit/circuitpython/blob/main/ports/stm/common-hal/busio/I2C.c#L231

fwiw, it doesn't look like the API call being used supports it; I'll dig in a bit more later:
[image]

https://www.st.com/resource/en/user_manual/dm00105879-description-of-stm32f4-hal-and-ll-drivers-stmicroelectronics.pdf

hierophect, 2020-11-24:
Looks like this is a consequence of using the blocking I2C Master Mode. We'll need to replace it with the interrupt mode for repeated starts:
[image]

hierophect, 2020-11-24:
Other option is to use HAL_I2C_Mem_Read, which does this repeated start automatically, but I'm not sure how flexible it is compared to the IT version.
[Relevant](https://community.st.com/s/question/0D50X00009XkgOYSAZ/support-for-repeated-start-in-stm32-hal-i2c-library) [posts](https://community.st.com/s/question/0D50X0000ADCf1mSQD/stm32f7-hal-i2c-repeated-start-command-for-the-max34409-current-sensing)

siddacious, 2020-11-26:
@hierophect it looks like `HAL_I2C_Mem_Read` uses a `uint16_t` for the address to read, implying that it can only send one or two bytes before doing the repeat start and read. While this would be useful for most use cases, it would imply putting a limit on the size of the out/send buffer for `write_then_readinto` which would be a change to the API.

Off the top of my head I can't think of a case where I've needed to write more than 2 bytes to set up a read, so it would probably work for most cases (until it doesn't).

That said, if the interrupt based approach isn't burdensome, it would be preferable

PR #4169 (2021-02-18): STM32: Fix I2C repeated start by converting to IT mode
The existing implementation of I2C does not currently recognize requests for a repeated start condition, which caused reads to fail on some chips. Since only the "IT" interrupt-driven versions of the ST HAL support this feature, fixing this issue also required implementing the usual infrastructure for ST interrupts, matching other modules like UART and PulseIn.

@tannewt I didn't really touch the base implementation here, but if you think we'd benefit from moving regular I2C operations to the interrupt mode (non-blocking) I can add in the  [... cut]
closed 2021-02-18 by hierophect