adafruit/circuitpython #3742: Getting pystack exhausted when trying to run

nbJosh, 2020-11-23:
Started getting pystack exhausted on 6.x beta buds,  was able to make some changes. and got my project working,  but now it's every time,  and no matter what I do,  I can't get my current source to run.   and I don't think I made any changes that would effect stack usage.   

Unfortunately I can not share the project.    but there was a note in early 6.x alphas about "Letting us know" if we get pystack exhausted,  and,  well,  I am getting that in spades now.

Thanks

tannewt, 2020-11-24:
Please increase CIRCUITPY_PYSTACK_SIZE in py/circuitpy_mpconfig.h until it works for you and let us know. We should be able to increase it.

What microcontroller are you using?

nbJosh, 2020-11-24:
I'm on a MatrixPortal

tannewt, 2020-11-24:
Please let us know what value works for you.

tannewt, 2020-12-29:
Closing because we never heard of a better number.

closed 2020-12-29 by tannewt

jay0lee, 2020-12-29:
Just wanted to point out that I was seeing this exact error when using the adafruit_jwt library and having a wrongly formatted value set for private_key (string instead of set). That library should probably be doing type checking on the value but the point is, this error may be a bad configuration elsewhere as opposed to the value of CIRCUITPY_PYSTACK_SIZE being to low.

a-t-0, 2022-01-30:
In case others look at this, I got this error when I tried to add a `KC.ASTERISK` key in my `keyboard.keymap = [ [...`. Removing that element from the keymatrix resolved the error.