adafruit/circuitpython #3743: ESP32S2 hangs when interrupting running code that includes I2C object

skieast, 2020-11-23:
Using the current (as of now) main  App version:      6.1.0-beta.1-14-g16fe2de70 I can running code to hang after interrupting, 
The code only instantiates an I2C object then loops forever printing something. 
Interrupted with CTRL-C works 1st time
Interrupted again with CTRL-C hangs. 
This was tested with both an Espressif Saola WROVER and Unexpectedmaker FeatherS2
```python
import wifi
import time
import board
from busio import I2C,SPI

# i2c = I2C(board.SCL, board.SDA)
i2c = I2C(board.IO2, board.IO1)  
# spi = SPI(board.SCK, MISO=board.MISO)

while True:
    print("Looping for fun\n")
    time.sleep(1)
```
The code was compiled with DEBUG=1 which outputs some wifi debug info.  Nothing interesting (to me appeared)
I have attached the log from restart to hang. 
[log.txt](https://github.com/adafruit/circuitpython/files/5585492/log.txt)
The last output before the hang was:
```text
I (15918) wifi:enable tsf
I (15918) wifi:mode : sta (7c:df:a1:00:49:fe)
I (27298) wifi:flush txq
I (27298) wifi:stop sw txq
I (27298) wifi:lmac stop hw txq
I (27298) wifi:Deinit lldesc rx mblock:4
I (27308) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pull                                       down: 0| Intr:0
```

gamblor21, 2020-11-23:
I was able to reproduce on my FeatherS2 (beta 6.1 beta 1).

If I comment out import wifi it continues to run perfectly. If I comment out of the I2C lines it works fine. When I put it all together then on the second Ctrl-C it locks up.

I also saw this:
```
Looping for fun

soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
```
I also saw the "soft reboot" and restart come up randomly as well when I hit Ctrl-C.

jepler, 2020-11-24:
Is wifi necessary to produce this problem?

gamblor21, 2020-11-24:
If I comment out the `import wifi` I cannot cause this to reproduce.

Before any reboots, if I uncomment out `import wifi` it will lock up the next Ctrl-C.

So it appears the `import wifi` is necessary to cause the issue to occur.

skieast, 2020-11-25:
Yes, same.
Wifi and I2C hangs (in my case on the 2nd CTRL-C
Leave either out it doesn't hang.  Nothing is actually being done to the
wifi or i2c in my code besides the import and instantiate an i2c object. I
assume wifi is instantiated on import.

tannewt, 2020-11-25:
Yes, the wifi subsystem is initialized on import. It's done here: https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/common-hal/wifi/__init__.c#L94-L134

skieast, 2020-11-26:
As a probably obvious observation removing all the I2C hardware/software init and deint removes the crash as well. I was a bit bored yesterday so did that to confirm. Have a great safe Thanksgiving to those celebrating that. Here in Canada I went skiing :)

skieast, 2020-11-28:
I have eliminated the crash in my test code. I have to do a bit more dissection to verify which change actually fixed things.

tannewt, 2020-11-30:
@skieast Could you fill us in on where you are at? We'd like to get this fixed this week. Jeff or I can pick up where you left off if needed.

skieast, 2020-12-01:
> 
> 
> @skieast Could you fill us in on where you are at? We'd like to get this fixed this week. Jeff or I can pick up where you left off if needed.

I thought i was getting somewhere but didnt happen.  So not at any state really. I had put it aside for the weekend but will take another look tomorrow.

skieast, 2020-12-05:
So after a bit of experimenting I found that by only doing the i2c_driver_install once, never doing an i2c_driver_delete then everything works. Now I haven't dug into the depths of the IDF to attempt to discover why. But it works. So a clue.  I'm not sure if actually not deleting the driver when i2c_reset is called is an issue. 
My current branch is here [https://github.com/skieast/circuitpython/tree/fix_i2c](https://github.com/skieast/circuitpython/tree/fix_i2c)
I also was working on some semaphore changes as I don't think the current implementation was correct. As my knowledge of FreeRTOS is a week old I could be wrong.  Should have made a separate  branch.  Hindsight is perfect.

microDev1, 2020-12-07:
Strange... I tested this with i2cperipheral which basically has same mechanics for init & deinit as busio.i2c and didn't have this issue. At the same time this issue appears when I switch to busio.i2c.

skieast, 2020-12-07:
> 
> 
> Strange... I tested this with i2cperipheral which basically has same mechanics for init & deinit as busio.i2c and didn't have this issue. At the same time this issue appears when I switch to busio.i2c.

Is that code available somewhere?   Thanks

microDev1, 2020-12-07:
I have a open PR for i2cperipheral > 3768. The build artifacts are available [here](https://github.com/adafruit/circuitpython/actions/runs/388294352).

Test Code:
```python
import wifi
import time
import board
from i2cperipheral import I2CPeripheral

i2c = I2CPeripheral(board.SCL, board.SDA, (0x40, 0x41))

while True:
    print("Looping for fun\n")
    time.sleep(1)
```

closed 2020-12-09 by tannewt