adafruit/circuitpython #3746: BLE HID Keyboard not working in CircuitPython 6.0

falkenad, 2020-11-23:
I have both a NRF52840 Feather and an ItsyBitsy NRF52840 Express, and I'm trying to make the BLE HID Keyboard example here work:

[](https://learn.adafruit.com/ble-hid-keyboard-buttons-with-circuitpython/ble-keyboard-buttons-libraries-and-code)

This does not seem to work on CircuitPython 6.0. I have updated both the Itsybitsy and the Feather to CircuitPython 6.0, and installed the 6.0 libraries as listed (adafruit_ble,adafruit_bus_device,adafruit_hid,neopixel.mpy,simpleio.mpy).

Any ideas what might need to change?
Seems this was made for an older version of Circuitpython, since it was released in January 2020, and I'm trying this on 6.0.0 in 11/20. I can't seem to get any of my Windows PCs, Android, or iOS devices to recognize either device when trying to pair it.

falkenad, 2020-11-24:
I'll add that it works just fine with the Feather on 5.0.0. It does not seem to work at all on the Itsybitsy NRF52840 on 5.x despite the obvious similarities between the Feather and Itsybitsy.

tannewt, 2020-11-24:
It works with 5.3.1 on both right?

falkenad, 2020-11-24:
Correct, I don't seem to have any issues with 5.3.x. I have tried 5.3.x with both the Feather NRF52840 Express and the Itsybitsy NRF52840 Express without problems. When I tried the example code with 6.0.0 it seemed to fail and not advertise, whereas with 5.3.1 I can immediately see "Circuitpython HID" appear in the device list of a host device. I haven't  yet stepped through line by line to see if it fails anywhere.

falkenad, 2020-11-24:
I'm not sure what the easiest way to step through is, if there's  a way to feed it through the REPL line by line easily.

tannewt, 2020-11-25:
> I'm not sure what the easiest way to step through is, if there's a way to feed it through the REPL line by line easily.

There isn't. If you have a JLink or similar debugger you can do it from the C side of CircuitPython. Let me know if you'd like guidance on how to dig into it.

We should have time to look into this once we get the S2 a bit more settled.

tannewt, 2020-12-29:
@falkenad Please try 6.0.1. We believe this is fixed.

dhalbert, 2020-12-29:
Fixed by #3861. I thought the issue was linked.

PR #3861 (2020-12-22): BLE fixes
(Same as #3848, but applied against `6.0.x` instead of `main`; MagTag name fix not included because board is not in 6.0.x)

- Characteristic and descriptor `max_length`s can be zero. Fixes #3746.
- Assure that `start_scan()` timeout is either zero or >= interval. This is a guess for this fix, but it coincides with working values reported in #3826. Fixes #3826.
- Assure that `start_scan()` max timeout is not too large for a `uint16_t` when passed to nRF SD. Fixes #3340.
- Fixes small advertising interval values. Fix check and make sure it's at least 0.02001, to allow for rounding issu [... cut]
closed 2020-12-29 by dhalbert