adafruit/circuitpython #3748: dir(cp) doesn't work

ajs256, 2020-11-24:
```
>>> from adafruit_circuitplayground import cp
>>> dir(cp)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_circuitplayground/express.py", line 82, in _unsupported
NotImplementedError: This feature is not supported on Circuit Playground Express.
>>> import math # to check that dir() works on something else
>>> dir(math)
['__class__', '__name__', 'acos', 'asin', 'atan', 'atan2', 'ceil', 'copysign', 'cos', 'degrees', 'e', 'exp', 'fabs', 'floor', 'fmod', 'frexp', 'isfinite', 'isinf', 'isnan', 'ldexp', 'log', 'modf', 'pi', 'pow', 'radians', 'sin', 'sqrt', 'tan', 'trunc']
```
I am not sure why this is happening. `cp` imports correctly, but `dir(cp)` doesn't work.

DavePutz, 2020-11-28:
It looks like this is being caused by PR #86 to the adafruit/Adafruit_CircuitPython_CircuitPlayground library . By adding a property of "_unsupported" to the Express class, which raises a NotImplementedError when accessed, it causes any process (like 'dir(cp)) which accesses that property to stacktrace. I guess the question is whether it is more valuable to have a good message when someone tries to access a non-existent feature or to have commands like dir() work.

dhalbert, 2021-02-11:
Closing in favor of dupe #4171, which has a bit more discussion.

closed 2021-02-11 by dhalbert