adafruit/circuitpython #3750: ESP32S2 does not properly reset on microcontroller.reset()

Neradoc, 2020-11-24:
On the ESP32S2 port, microcontroller.reset() makes the REPL unresponsive and does seemingly nothing else. The drive remains available. Pressing the reset button is necessary to recover.

dhalbert, 2020-11-24:
I can check on this as part of the sleep/alarm work.

hierophect, 2020-11-24:
Is it possible that this is being overzealous about resetting modules, and is messing with pins related to the REPL?

tannewt, 2020-11-24:
The REPL pins are USB pins. The drive wouldn't work if USB was broken.

microDev1, 2020-12-07:
Is this implemented? REPL unresponsive can be explained by infinite loop in `common_hal_mcu_reset()`.

I think following should do. The reset reason in this case is `ESP_RST_SW`. Should I make a PR?
```c
void common_hal_mcu_reset(void) {
    filesystem_flush(); //TODO: implement as part of flash improvements
    esp_restart();
}
```

dhalbert, 2020-12-08:
> I think following should do. The reset reason in this case is `ESP_RST_SW`. Should I make a PR?

Sure, go ahead, thanks! I forgot to do this before handing the sleep code back to @tannewt.

dhalbert, 2020-12-15:
Fixed by #3804. Thanks @microDev1!

closed 2020-12-15 by dhalbert