adafruit/circuitpython #3751: The firmware of esp32-s2 is normal by nano official test. Is there a problem with the board?

60999, 2020-11-24:
I'm sorry to disturb you. I use nanoesp32-s2 / firmware（ https://github.com/wuxx/nanoESP32-S2/blob/master/firmware/flash_ image_ cdc_ msc.bin ）, write it to 0x00, connect gpio19 and gpio20 (the schematic diagram is the same as nano, add resistance in the middle), insert USB2.0 interface into the computer, and successfully display the small U disk.

Then we use adafruit circuit Python muselab_ nanoesp32_ s2-zh_ Latn_ Pinyin-6.0.0.bin, write to 0x00

Serial terminal display:

ESP- ROM:esp32s2-rc4-20191025

Build:Oct 25 two thousand and nineteen

rst:0x1 (POWERON), boot:0x8 (SPI_ FAST_ FLASH_ BOOT)

SPIWP:0xee

mode:DIO , clock div:2

load:0x3ffe6100 , len:0x8

load:0x3ffe6108 , len:0x18a0

load:0x4004c000 , len:0x930

load:0x40050000 , len:0x2cc8

entry 0x4004c1b4

What is the reason for this? How can I fix this problem?

60999, 2020-11-24:
By the way, my motherboard is the esp-12k development board nodemocu-32-s2

tannewt, 2020-11-24:
What is the issue? That UART output looks ok.

tannewt, 2020-11-25:
I don't know anything about uPyCraft. If it is running successfully, you should see a drive appear. You can build a debug version of CircuitPython and see what UART logging you get with that.

60999, 2020-11-27:
I confirm that I have seen the drive appearance, but only these contents are displayed in the terminal environment. I cannot enter the development environment starting with ">>>"

dingramax, 2020-11-30:
Try connecting with putty. On the board I'm using currently the Saola M1 all i have to do is put in the com port I don't need to put in the baud rate or anything the board figures it out ymmv

60999, 2020-12-01:
The same problem occurs with another piece of nodemocu-32-s2.

dhalbert, 2020-12-01:
@60999 Please try to connect with another terminal program, such as putty or Tera Term, as suggested by @dingramax.

I am confused about which board firmware you're using: you say you're using `muselab_nanoesp32_s2`, but your board is "nodemocu-32-s2", so they don't match. Is it "mocu" or "mcu"? Could you post a pointer to that board? It's possible the pin connections are different.

60999, 2020-12-01:
> Try connecting with putty. On the board I'm using currently the Saola M1 all i have to do is put in the com port I don't need to put in the baud rate or anything the board figures it out ymmv

I used putty to get the same results.
Is the firmware incompatible? It's also an esp32-s2 chip.

60999, 2020-12-01:
> @60999 Please try to connect with another terminal program, such as putty or Tera Term, as suggested by @dingramax.
>
> I am confused about which board firmware you're using: you say you're using [...]

I use esp32 S2 chip esp-12k development board: nodemocu-32-s2, burning Firmware of 
 muselab_ nanoesp32_ S2 development board.

60999, 2020-12-03:
When I use esp32 S2 chip nanoESP32-S2 development board, burning Firmware of
nanoESP32-S2 development board.

`ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x18a0
load:0x4004c000,len:0x930
load:0x40050000,len:0x2cc8
entry 0x4004c1b4
`
That's the same display.

60999, 2020-12-03:
Sorry ,my mistake. It must use another usb port.Thanks for help.

closed 2020-12-03 by 60999