adafruit/circuitpython #3753: On-board NeoPixels max brightness when brightness is set to 0

kattni, 2020-11-24:
Tested on MagTag, PyBadge, CPX and QT Py. The issue is not present on QT Py which requires `adafruit_pypixelbuf`. It is present on all other boards, which may narrow it down to an issue in `_pixelbuf`. 

Used simple code that uses `board.NEOPIXEL` to turn the pixels red, and then tested brightness:

- Setting `brightness = 0` makes the on-board NeoPixels max brightness. They are the appropriate color when this occurs (as in they are not getting incorrect color info).

- Setting `brightness = 0.1` makes them dim.

- Setting `brightness = 0.0` shuts them off, though that's likely because it thinks it's something like `0.000000001` and they're not actually "off" but the LEDs or human eyes can't handle that level of dimness.

brrzap, 2021-01-01:
i can reproduce this using a custom build on a QT Py that adds the native `_pixelbuf`.  it does not occur using `brightness=0` in the constructor:

```py
import board
import neopixel
np = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0)
# no change visible yet
np[0] = (100,0,0)  
# now turned off
np.brightness = 0
# back to full on
```

looking in `shared-bindings/_pixelbuf/PixelBuf.c`, the [constructor calls `mp_obj_get_float()`](https://github.com/adafruit/circuitpython/blob/d0bc5e4e5737f330d87177cd4c88507a176e2c87/shared-bindings/_pixelbuf/PixelBuf.c#L104), while the [setter uses `mp_obj_float_get()`](https://github.com/adafruit/circuitpython/blob/d0bc5e4e5737f330d87177cd4c88507a176e2c87/shared-bindings/_pixelbuf/PixelBuf.c#L188).  this skips [`..._get_float()`'s extra error-checking](https://github.com/adafruit/circuitpython/blob/d0bc5e4e5737f330d87177cd4c88507a176e2c87/py/obj.c#L332) that seems to catch this case for the constructor.  i can't follow the [bitmath in `mp_obj_float_get()`](https://github.com/adafruit/circuitpython/blob/d0bc5e4e5737f330d87177cd4c88507a176e2c87/py/obj.h#L150) to prove the fault, but testing shows that the setter takes the `brightness > 1` fork after it's called.

changing the setter to use `mp_obj_get_float()` instead of `mp_obj_float_get()` does seem to fix the error, but it will add some cycles.  changing the bitmath might be another option but i don't have any suggestions as to how.

ladyada, 2021-01-01:
@brrzap good catch, wanna submit a PR for that setter change? its ok if it adds some cycles - better to have it right :)

PR #3917 (2021-01-02): pixelbuf: error check in brightness setter
sync brightness setter's float cast to that used in constructor for error checking. fixes #3753.
commit df4c028d (2021-01-02): _pixelbuf: error check in brightness setter (fix #3753)
closed 2021-01-02 by dhalbert