adafruit/circuitpython #3762: (Large numbers of) Neopixel  too slow - Again

jammerxd, 2020-11-25:
Hi,

First off, thanks for an awesome library!

I have a string of 300 LEDs that I want to update in sync with music using a chase down the line effect, however, the "show" method is very slow and causes the system to fall behind when processing the music.

I'll share my example code that's running it. I'm currently using a C# application to send a 7 band spectrum from 0-100 for volume, then python is decoding that set of bytes into the proper numbers and when the data is completely received, then calls to update the LED strips and the pixels. If I only update the pixels and do not show them, the code runs fine, however, as soon as I implement a call to pixels.show(), the whole system lags behind in processing the packets because pixels.show() takes so long to finish it's work. I've also investigated the higher end SimpleNeoPixel demo written for arduino and wonder why I can't get that kind of animation going, but be reactive to music. Please let me know where the code can be improved to be reactive to the data being received over the network.(Data is fed every 8ms and the PWM/I2C bus has 0 issues keeping up)
 
Everything is able to keep up if I just want 1 color, but as soon as I want "zones" or different effects on the addressable LED strip, the whole system slows down.

https://pastebin.com/5X8gFtV5

dhalbert, 2020-11-25:
Hi, this is not really a bug, so it's more appropriate to bring it up in https://forums.adafruit.com. Could you post there, with more details?

Are you running this on a microcontroller board or on an RPi or similar? I'm not sure, based on the network code. I don't see a main program here, either.

Depending on the board, there is a native implementation of some of `NeoPixel` in an internal module called `_pixelbuf`. On small boards such as Trinket, that code is not native, but is in Python, which makes it slower. So in your post, say what kind of board you're using.

Using brightness other than 1.0 causes a slowdown because the data must be copied.

closed 2020-11-25 by dhalbert

PureWeen, 2021-07-25:
> Depending on the board, there is a native implementation of some of NeoPixel in an internal module called _pixelbuf. On small boards such as Trinket, that code is not native, but is in Python, which makes it slower. So in your post, say what kind of board you're using.

Is there a way to know the boards that it's native on vs in Python?

todbot, 2021-07-25:
> Is there a way to know the boards that it's native on vs in Python?
Yep! Search for `pixelbuf` in the Module Support Matrix to get a list of boards that use the (newly renamed) built-in C-based `adafruit_pixelbuf` module: https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html?filter=adafruit_pixelbuf