adafruit/circuitpython #3763: Teensy 4 busio not working in Circuitpython 6.0.0 with lib 6.x bundle from 20201007

grjohn14, 2020-11-25:
Trying to mount an sd card or just a small test on bno055 as below both report:  OSError: [Errno 5] Input/output error

**code used:**
import time
import board
import busio
import adafruit_bno055
 
# Use these lines for I2C
i2c = busio.I2C(board.SCL, board.SDA)
sensor = adafruit_bno055.BNO055_I2C(i2c)
 
# User these lines for UART
# uart = busio.UART(board.TX, board.RX)
# sensor = adafruit_bno055.BNO055_UART(uart)
 
while True:
    print("Temperature: {} degrees C".format(sensor.temperature))
    print("Accelerometer (m/s^2): {}".format(sensor.acceleration))
    print("Magnetometer (microteslas): {}".format(sensor.magnetic))
    print("Gyroscope (rad/sec): {}".format(sensor.gyro))
    print("Euler angle: {}".format(sensor.euler))
    print("Quaternion: {}".format(sensor.quaternion))
    print("Linear acceleration (m/s^2): {}".format(sensor.linear_acceleration))
    print("Gravity (m/s^2): {}".format(sensor.gravity))
    print()
 
    time.sleep(1)

**Full error:**
Traceback (most recent call last):
  File "code.py", line 17, in <module>
  File "adafruit_bno055.py", line 385, in magnetic
  File "adafruit_bno055.py", line 141, in __get__
  File "adafruit_register/i2c_struct.py", line 59, in __get__
  File "adafruit_register/i2c_struct.py", line 59, in __get__
  File "adafruit_bus_device/i2c_device.py", line 143, in write_then_readinto
OSError: [Errno 5] Input/output error

**Could be related to a similar problem with ESP32 -S2:**
https://github.com/adafruit/circuitpython/issues/3396

Any input greatly appreciated.

 - George

jepler, 2021-03-24:
We believe the SPI "status 400" problem has been fixed by #4048.  However, we will use this issue to track the reported I2C problem.

jepler, 2021-03-26:
Double checked that sdcardio still works after #4487.

ladyada, 2021-03-26:
this issue seems specific to the BNO055 which is a wretched chip, may be related to clock stretching

dhalbert, 2023-04-11:
The BNO055, and some other BNOxx sensors, have bugs in their I2C implementations. Some chip families handle these better than others. We are going to attempt to work around at least some of these in the CircuitPython libraries, and will add more warnings to the sensor guides. I'll close this now since it does not reflect a bug in our mimxrt10xx I2C implementation itself.

closed 2023-04-11 by dhalbert

grjohn14, 2023-04-13:
Lol.. What is completed? Nothing.. Excellent work sir!