adafruit/circuitpython #3765: Missing definitions in board  ESP32S2

mpechner, 2020-11-27:
Trying to run this project
https://learn.adafruit.com/pyportal-iot-plant-monitor-with-aws-iot-and-circuitpython?view=all
In this code:  esp32spi_simpletest.py

What pins should I use? The uncommented definitions are not defined.

code snippet:
esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)

# If you have an AirLift Shield:
# esp32_cs = DigitalInOut(board.D10)
# esp32_ready = DigitalInOut(board.D7)
# esp32_reset = DigitalInOut(board.D5)

# If you have an AirLift Featherwing or ItsyBitsy Airlift:
# esp32_cs = DigitalInOut(board.D13)
# esp32_ready = DigitalInOut(board.D11)
# esp32_reset = DigitalInOut(board.D12)

# If you have an externally connected ESP32:
# NOTE: You may need to change the pins to reflect your wiring
# esp32_cs = DigitalInOut(board.D9)
# esp32_ready = DigitalInOut(board.D10)
# esp32_reset = DigitalInOut(board.D5)

anecdata, 2020-11-27:
Those pin definitions are not for ESP32-S2, they are for ESP32 when used as a wi-fi co-processor. You'll want code more like this for the ESP32-S2: [https://learn.adafruit.com/adafruit-metro-esp32-s2/internet-connect](https://learn.adafruit.com/adafruit-metro-esp32-s2/internet-connect)

siddacious, 2020-11-27:
@mpechner, @anecdata is correct. Please try his suggestion and if you have issues we can help with support in the adafruit forums: https://forums.adafruit.com

closed 2020-11-27 by siddacious