adafruit/circuitpython #3772: crash in wifi_module.py connecting to wifi

endico, 2020-11-29:
I'm using Adafruit CircuitPython 6.1.0-beta.1 on 2020-11-20; MagTag with ESP32S2
the adafruit-circuitpython-bundle-6.x-mpy-20201126 library on a MagTag.

I'm getting intermittent crashes when connecting to wifi. For most of the day yesterday it happened once in a while and if I re-saved in Mu, the connection would work on the second try. Then last night it would not stop crashing and I gave up and went to bed. This morning I'm trying to write the bug report and now it won't crash at all. Maybe my eero was throttling the MagTag after so many connections in one day? I poked around in the Eero app and didn't notice any problems at that end. It usually shows the MagTag with a weak signal but it was stronger than normal.

Traceback (most recent call last):
  File "code.py", line 19, in <module>
  File "adafruit_magtag/magtag.py", line 89, in init
  File "adafruit_magtag/magtag.py", line 407, in url
  File "adafruit_magtag/network.py", line 309, in connect
  File "adafruit_magtag/network.py", line 304, in connect
  File "adafruit_magtag/wifi_module.py", line 67, in connect
ConnectionError: No network with that ssid

tannewt, 2020-11-30:
I'm not sure how to help. It definitely sounds related to your network setup. Let me know if you'd like tips for debugging it. A start may be repeatedly scanning and seeing how often you don't see your ssid in the list.

BennyE, 2020-12-02:
Can you share your code @endico? How many Eero APs do you have? According to [this article](https://support.eero.com/hc/en-us/articles/115001613323-How-do-I-block-unwanted-devices-on-my-network-) the device would show up as "Blocked" on your home screen (if this would be an issue).

endico, 2020-12-06:
I'm using just one Eero. The magtag doesn't show up as being blocked in the eero app. Usually if there's a crash I can just hit save in Mu once or twice and it will finally work.

Today its crashing with this:
Traceback (most recent call last):
  File "code.py", line 18, in <module>
  File "adafruit_magtag/magtag.py", line 89, in __init__
  File "adafruit_magtag/magtag.py", line 407, in url
  File "adafruit_magtag/network.py", line 309, in connect
  File "adafruit_magtag/network.py", line 304, in connect
  File "adafruit_magtag/wifi_module.py", line 67, in connect
ConnectionError: Unknown failure

Its doing this with the cheerlights demo whose link I can no longer find

# SPDX-FileCopyrightText: 2017 Scott Shawcroft, written for Adafruit Industries
#
# SPDX-License-Identifier: Unlicense
import time
import board
from adafruit_magtag.magtag import MagTag
import neopixel

external_pixels = neopixel.NeoPixel(board.D10, 30, brightness=0.3)

# Set up where we'll be fetching data from
DATA_SOURCE = "http://api.thingspeak.com/channels/1417/field/2/last.json"
COLOR_LOCATION = ['field2']
DATE_LOCATION = ['created_at']

magtag = MagTag(
    url=DATA_SOURCE,
    json_path=(COLOR_LOCATION, DATE_LOCATION),
)
magtag.network.connect()

# Color
magtag.add_text(
    text_font="Arial-Bold-12.bdf",
    text_position=(10, 15),
    text_transform=lambda x: "New Color {}".format(x),
)
# datestamp
magtag.add_text(
    text_font="Arial-Bold-12.bdf",
    text_position=(10, 35),
    text_transform=lambda x: "Updated on: {}".format(x),
)

timestamp = None
while True:
    if not timestamp or ((time.monotonic() - timestamp) > 10):  # once every 10 seconds...
        try:
            value = magtag.fetch()
            print("Response is", value)
            magtag.peripherals.neopixel_disable = False
            color = int(value[0][1:], 16)
            magtag.peripherals.neopixels.fill(color)
            external_pixels.fill(color)
            timestamp = time.monotonic()
        except (ValueError, RuntimeError) as e:
            print("Some error occured, retrying! -", e)
    time.sleep(1)

endico, 2020-12-06:
@tannewt "A start may be repeatedly scanning and seeing how often you don't see your ssid in the list."

I didn't try this tonight but when I scanned a couple of weeks ago I noticed that it only listed about 5 SSIDs and often my SSID wasn't listed but if I kept trying it would eventually show up. I live in an apartment building and there's a lot of wifi around. My mac lists about 45 SSIDs and 18 of them have strong signals. So maybe my SSID doesn't show up because its got too much competition.

endico, 2020-12-06:
OK, I'm running this code now and its listing about 10 SSIDs including mine. The first time I tried it there was a long wait but eventually it connected. After that, I've rerun it a few times and keep getting 
ConnectionError: Unknown failure. Eventually I was able to reconnect and when I reloaded the magtag again right away it got the unknown failure again.

```
import wifi
for network in wifi.radio.start_scanning_networks():
    print("\t'%s'\t\tRSSI: %d\tChannel: %d" % (network.ssid, network.rssi, network.channel))
wifi.radio.stop_scanning_networks()
print("connecting...")
wifi.radio.connect(ssid="mysssid", password="supersecret")
print("connected.")
```

BennyE, 2020-12-06:
Hi @endico,

I had best results by *not* calling wifi.radio.stop_scanning_networks(), as I started with a similar code like you shared (a few weeks ago).
I know it may sound a bit awkward, but could you have a look if the code like this would give you a better result?
When I had trouble, I also started to no longer specify ssid="" and password="" but just give them as positional arguments.

```python
import wifi
import time
for network in wifi.radio.start_scanning_networks():
    print("\t'%s'\t\tRSSI: %d\tChannel: %d" % (network.ssid, network.rssi, network.channel))
#wifi.radio.stop_scanning_networks()
print("connecting...")
wifi.radio.connect(ssid="mysssid", password="supersecret")
# please also try
#wifi.radio.connect("mysssid", "supersecret")
print("connected.")
while True:
    print("looping for fun")
    time.sleep(3)
```

Your Wi-Fi AP should be on channel 1, 6 or 11. These are the only valid choices, due to the channel interference/overlap that the 2.4 GHz frequency spectrum has.

Does your SSID name include a whitespace e.g. "My Wi-Fi" instead of "MyWi-Fi"?
Is there, by any chance, the same SSID around twice e.g. by another EERO customer?

endico, 2020-12-06:
This morning (8:30am) everything is working fine as it often does in the morning. It seems like the problem happens more often at night. I'll try again later. I'm not sure what difference time of day makes. I guess there's more WI-FI activity at night. The list of nearby APs seems to be about the same.

Yes, my AP is on channel one. I actually have two SSIDs. The guest network contains a space and the main one doesn't. I've never been able to connect to the guest network with the MagTag even though it shows up in my scans. 

There don't seem to be any other APs using my SSID with the possible exception of the guest network. In the past I've noticed that when I connect via the guest credentials that I showed up as being connected to the main network. That doesn't seem to happen now but maybe something funny is still going on.

anecdata, 2020-12-17:
The Wi-Fi Radio module could perhaps react to more disconnection [reasons](https://github.com/jepler/esp-idf/blob/d06744f5efc382c61cbad8758107cec308feef09/components/esp_wifi/include/esp_wifi_types.h#L64) to make the message more specific in [connect](https://github.com/adafruit/circuitpython/blob/68dd4b2b37d35218e306a0de2efb659cbd505fb6/ports/esp32s2/common-hal/wifi/Radio.c#L171).

The event handler is aware of certain common [reasons](https://github.com/adafruit/circuitpython/blob/68dd4b2b37d35218e306a0de2efb659cbd505fb6/ports/esp32s2/common-hal/wifi/__init__.c#L62) for disconnection, and will automatically re-connect in those cases.

anecdata, 2020-12-17:
For now, if you haven't already tried these things:
• reboot the router and MagTag when the issue is occurring and see if it improves
• see if you can re-position the MagTag so there's a better path from its antenna to the router
• add `ConnectionError` to your try/except
interference could be a complicating factor, depending on 2.4MHz device density at your location

endico, 2020-12-17:
Thanks, I added ConnectionError to my try/except and it still didn't catch the connection error and proceeded to crash with a Connection error.

I guess I'll try using the wifi Radio module. I think that might solve my display problems too.

endico, 2020-12-19:
Update: Yesterday I finished my project to the point where I could plug in the battery and let it run overnight sitting on the fridge near the router. It managed to last the night without crashing from a network error.

Today I updated CP to the latest daily build and am still seeing intermittent connection errors from the computer in the back room. Sometimes "unknown error" and sometimes "No network with that ssid". When this happens, I re-save in Mu and it usually works on the first or second retry.

d-c-d, 2020-12-19:
@endico, here is what I needed to do to get the proper retries to work and restart things correctly ( from the christmas countdown for context )

`
#  create MagTag and connect to network
try:
    magtag = MagTag()
    magtag.network.connect()
except (ConnectionError, ValueError, RuntimeError) as e:
    print("*** MagTag(), Some error occured, retrying! -", e)
    # Exit program and restart in 1 seconds.
    magtag.exit_and_deep_sleep(1)

`

endico, 2020-12-20:
Thanks, David. I fixed my code per d-c-d's suggestion and I'm no no longer crashing on connection errors. I'm still getting plenty of 'No network with that ssid' and Unknown failures but now the code is retrying, and eventually connecting.

endico, 2020-12-26:
I thought I had made a lot of progress when I figured out that my 'Unknown failures' are of type OSError, not ConnectionError. Except "No network with that ssid" is also an OSError. I'll post my test code later.

endico, 2020-12-27:
My ConnectionError crashes were due to me not catching the errors properly. So on that basis you might want to close out this issue. On the other hand, I am still getting **unknown error** and I think you were trying to fix those. I still get the errors, but I'm not crashing any more.  I created this script to count connection errors and log them on Adafruit IO. See the README for more details.  
https://io.adafruit.com/endico/feeds/failures

I have a close neighbor who is a gamer and I thought maybe some of his gear was creating interference but he was away yesterday and I was still getting errors (more than today). I guess I'll let this run for a week or two and see if any patterns emerge.

https://github.com/endico/Magtag-Battery-Test

Thundercatsffs, 2021-07-08:
Without having taken a look at your wifi myself id still go with the suggestion of it being a "wifi is wifi" sort of thing.
Im getting the same errors here and im in a very bad room, wifi wise. Its not uncommon to get request denials or getting similar issues when there are bad conenctions. Remember, the antenna on the esp32 isnt that big and since wifi is as it is...

However thank you all for the code, il try it myself and see if it works.

dhalbert, 2023-10-28:
Many things have changed since this was filed. I'm closing in favor of any newer issues that show similar problems.

closed 2023-10-28 by dhalbert