adafruit/circuitpython #3777: Support for OTA updates

microDev1, 2020-12-01:
The _esp32s2_ brings `WiFi` support and with that the possibility to have a mechanism for `OTA` updates. 

There are two domains for `OTA` update implementation:
- [ ] User Code Update
- [ ] CircuitPython Update 

### User Code Update
For this my plan is to expose the `circuitpy` drive as a network drive.

### CircuitPython Update
The plan here is to have the user code handle firmware grabbing and for the flashing part I agree with what @tannewt suggests.

>add an esp32s2 specific library that can write to the “other” OTA area and then switch to the alternate.

tannewt, 2020-12-01:
#3696 and #3454 could be helpful for user code updates. That way you don't need to overwrite the current, known good file.

microDev1, 2020-12-07:
I got the CircuitPython update working. The changes are in my [`ota-s2`](https://github.com/adafruit/circuitpython/compare/main...microDev1:ota-s2) branch.

With the current implementation the user provides buffer to be placed in `ota_0/1`. 

Question:
- Do we want the ability to update other flash partitions like `uf2`?
_Note: This could be risky as there is no rollback available for these partitions._

Todo enhancements: 
- Python library to handle firmware grabbing.
- Ability to flash firmware in continuous/discontinuous chunks.

Test Code:
- Build artifacts are [here](https://github.com/microDev1/circuitpython/actions/runs/405135038). 

```python
import ota
import ssl
import time
import wifi
import socketpool
import microcontroller
import adafruit_requests

# Firmware
url = "https://adafruit-circuit-python.s3.amazonaws.com/bin/microdev_micro_s2/en_US/adafruit-circuitpython-microdev_micro_s2-en_US-6.1.0-beta.2.bin"

# Start Address 0x10000 (refer: partitions.csv/makefile)
header = {}
header["Range"] = "bytes=65536-"

# Configure Wifi
for network in wifi.radio.start_scanning_networks():
    print(network, network.ssid, network.rssi, network.channel)

print()

wifi.radio.stop_scanning_networks()
wifi.radio.connect("ssid", "password")

# Setup Requests
pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

# Log Time
start_time = time.monotonic()

# Get Firmware
response = requests.get(url, headers=header)
print(response.status_code)
print(response._headers)

print()

# Update Firmware
print("Preparing Update")
ota.flash(response.content)
print("Update took:", time.monotonic() - start_time, "sec")

print()

print("done")

# Reset Microcontroller
microcontroller.reset()
```

tannewt, 2020-12-07:
>     * Do we want the ability to update other flash partitions like `uf2`?
>       _Note: This could be risky as there is no rollback available for these partitions._

Not now, let's start simple. :-)

microDev1, 2020-12-26:
CircuitPython update is implemented and User Code update can be done by a FTP server implementation in python layer as a library instead of implementing it in the core.

closed 2020-12-26 by microDev1

jamesjian007, 2021-01-01:
User Code update can be done by a FTP server implementation  @microDev1  Any example available ?

microDev1, 2021-01-06:
Hi @jamesjian007 There isn't a CircuitPython library to implement FTP server yet... I found a few MicroPython implementations which could be ported to `CPY`. Try searching for `micropython ftp` on github. :-)

ThreepE0, 2021-11-14:
Very interested in something like this; Managing the several devices scattered about would be magnitudes easier once this is implemented.

daniel-alsen, 2025-05-08:
Agree, a critical feature for real consumer applications. Please let me know if there is any progress on this. It would be great to be able to download both Circuitpython and user code over HTTPS and flash in smaller chunks, using Dualbank.