adafruit/circuitpython #3778: ESP32-C3 Support

Linjieqiang, 2020-12-01:
[image]

tannewt, 2020-12-01:
We won't be supporting the C3 because the USB device is a built in USB to UART and JTAG converter. It isn't a generic USB device peripheral. Sorry! Let Espressif know you'd like a full USB peripheral.

Linjieqiang, 2020-12-02:
That's a pity. Thanks. @tannewt

microDev1, 2021-08-24:
CircuitPython just got a `BLE` workflow, so `esp32c3` support is now possible.

closed 2021-09-28 by tannewt

s-light, 2021-11-03:
is there any other status for the port / support?
on [the docs](https://circuitpython.readthedocs.io/en/latest/ports/espressif/README.html#connecting-to-the-esp32-c3) its says only beta and
> BLE Connection:
> This feature is not yet available and currently under development.

on the [circuitpython.org/downloads](https://circuitpython.org/downloads) list at least the search does not find anything related to `C3` and in the filter `Processor family` there is only `esp32s2` available.

tannewt, 2021-11-04:
It is still very early. Looks like there are two C3 boards defined so far:
```
ports/espressif/boards/ai_thinker_esp32-c3s/mpconfigboard.mk
4:IDF_TARGET = esp32c3

ports/espressif/boards/microdev_micro_c3/mpconfigboard.mk
4:IDF_TARGET = esp32c3
```

s-light, 2021-11-05:
thanks for the update :-)

i have read that the cp bluethooth support relies on the WebBluethooth API that is currently only available in chrome...
so i think for me that is currently out of interest...
will have to dive deeper into this some time..

tannewt, 2021-11-05:
That's true for code.circuitpython.org. It is the best option for Android. We'll have separate iOS apps very soon for it.

s-light, 2021-11-05:
> dive deeper
→ i thought more in the direction of how can i use it with my linux laptop..
and in the end i probably will land back at the beginning:
wishing just this  wonderful simple USB-mass-storage-drive that just works for every one  ;-)

for the educational aspect i am on the line with you!
i just get to know that  in a growing number of schools here in germany pupils get a personal (lockeddown) ipad from school...  apple really did a great job on selling these things and looking the people into this thing :-(
so for this use case i also think an app will be the easiest way - as it fits in the system...