adafruit/circuitpython #3781: Grand Central - pwmio not functioning and no error being rasied

AdamCummick, 2020-12-02:
Running this code causes D19 to idle high and not oscillate. Removing D30 from pins_temp corrects the issue. No errors are flagged. D30 and D19 share TC4_WO1, but have different TCC peripherals.

```py
import board
import time
import pwmio

pins_temp = [ board.D30, board.D19]
pwm_pins = []
for pin in pins_temp:
    p = pwmio.PWMOut(pin, frequency=10, duty_cycle=0x7fff, variable_frequency=True)
    pwm_pins.append(p)

while True:
    print(10)
    for pin in pwm_pins:
        pin.frequency = 10
    time.sleep(3)
    print(20)
    for pin in pwm_pins:  
        pin.frequency = 20
    time.sleep(3)
```

tannewt, 2020-12-02:
What version of CircuitPython are you using?

AdamCummick, 2020-12-02:
I'm running 6.0.0

dhalbert, 2021-01-04:
Reproduced in 5.3.1 and later. The problem does not happen if `.frequency` is not assigned, and we just use the original frequency.

dhalbert, 2021-01-09:
@AdamCummick This will be fixed by #3960.

PR #3960 (2021-01-11): Look up TCC resolution as necessary
Fixes #3781.

`common_hal_pwmio_pwmout_set_frequency()` was assuming that all TCC's were 24-bit resolution, which is not correct.

Tested using variants of the original issue test program. It was confusing for a while because some other pins I tried worked. Looking at the register values showed that one `PER` value was truncated to 16 bits and one wasn't.
closed 2021-01-11 by tannewt