adafruit/circuitpython #3782: Add support for AT25DF641A SPI Flash (possibly other SPI Flash chips as requested)

skerr92, 2020-12-02:
I'm just adding this issue for reference in future PRs for additional CircuitPython SPI Flash support. 

I am making a feather that will utilize the AT25DF641A SPI Flash module. I discussed in the discord about adding support for it as there was other discussions about using larger SPI Flash. I will be testing with my CP Sapling M0 boards as well as the Qt Py once I get a few more Qt Py to test this with. 

I should note, if there are any other SPI Flash in the SOIC-8 that people want tested to be added support wise I can handle those requests. Just add the chip name in a comment on here and I can order them and test.

skerr92, 2020-12-05:
Going to test out support for the [MX25L25645G](https://www.macronix.com/Lists/Datasheet/Attachments/7858/MX25L25645G,%203V,%20256Mb,%20v1.9.pdf) 32MiB NOR SPI Flash as well. It appears that the 64MiB variant is already in the devices.h file, so the 32MiB seems to make sense.

dhalbert, 2020-12-21:
Fixed by #3824.

closed 2020-12-21 by dhalbert