adafruit/circuitpython #3784: esp32s2 build instructions missing a step?

kbanks-krobotics, 2020-12-03:
make V=2 BOARD=espressif_saola_1_wroom
GEN build-espressif_saola_1_wroom/genhdr/mpversion.h
python3 ../../py/makeversionhdr.py build-espressif_saola_1_wroom/genhdr/mpversion.h
IDF_PATH=/mnt/c/git/firialabs/circuitpython/ports/esp32s2/esp-idf cmake -S . -B build-espressif_saola_1_wroom/esp-idf -DSDKCONFIG=build-espressif_saola_1_wroom/esp-idf/sdkconfig -DSDKCONFIG_DEFAULTS="esp-idf-config/sdkconfig.defaults;esp-idf-config/sdkconfig-opt.defaults;esp-idf-config/sdkconfig-4MB.defaults;boards/espressif_saola_1_wroom/sdkconfig" -DCMAKE_TOOLCHAIN_FILE=/mnt/c/git/firialabs/circuitpython/ports/esp32s2/esp-idf/tools/cmake/toolchain-esp32s2.cmake -DIDF_TARGET=esp32s2 -GNinja
CMake Error: The source directory "/mnt/c/git/firialabs/circuitpython/ports/esp32s2/build-espressif_saola_1_wroom/esp-idf" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:253: recipe for target 'build-espressif_saola_1_wroom/esp-idf/config/sdkconfig.h' failed
make: *** [build-espressif_saola_1_wroom/esp-idf/config/sdkconfig.h] Error 1

The above is after a make clean.
Is there some intermediate command that should be done to fill in the build-<board name>/esp-idf directory with files (including a CMakeLists.txt)?

I believe my toolchain is basically correct, because I can build/flash/run examples directly from the ...ports/esp32s2/esp-idf tree (for example, Hello World).

Can someone who is currently able to build for esp32s2 check the published instructions for completeness?

Repo was grabbed on Monday, commit 5b3c930e384ef6440f0f7b5167bb54ea68a8be76

microDev1, 2020-12-03:
#3575 might be helpful...

kbanks-krobotics, 2020-12-08:
#3575 was of no help. For others who might get this same failure in the future, what worked for me was a fresh install of Ubuntu 20 LTS running under Virtual Box. I was only trying to build under Windows 10 WSL because that had worked previously for me with MicroPython. Some other things that may be helpful to others:
When you want to install ninja, it's actually called ninja-build.
The build process SEEMS to sometimes ask for "python" when it really wants python3. I made a link to force this.
GDB complained about libpython2.7.so.0.1 (name approx.) but that turns out to be apt installable by itself.
(sudo apt-get install libpython2.7)

Maybe there will be time in the future to revisit what specifically is failing when the build process fails with the empty esp-idf directory.

closed 2020-12-08 by kbanks-krobotics

davigupta, 2021-04-06:
I've ran into same error while building wrover on ubuntu 16.04 and while searching for solution on google came across this link. On further looking the issue is related to cmake version (3.5.1) in ubuntu 16.04 as the CMakeLists.txt in ports/esp32s2 of main branch needs 3.13. Using latest cmake 3.20 builds successfully.

dhalbert, 2021-04-06:
> The build process SEEMS to sometimes ask for "python" when it really wants python3. I made a link to force this.

On Ubuntu you can `sudo apt install python-is-python3` which sets up the links for you.

davigupta, 2021-04-06:
thanks for the tip. Just to add other observation that esp32s2/esp-idf.install.sh sets the virtualenv for python-2.7. After installing the new cmake version 20.2 and put it in my path list, build is successful for espressif_saola_1_wrover.

dhalbert, 2021-04-06:
if you can upgrade to 20.04 things will go more smoothly, though I can understand if that might be an issue.