adafruit/circuitpython #3785: MagTag battery monitor is not returning expected values

dhalbert, 2020-12-03:
There appears to be an issue with ADC calibration or operation. (Reported by @ladyada)

anecdata, 2020-12-03:
I think this should be 3.3, not 2.6:
[https://github.com/adafruit/Adafruit_CircuitPython_MagTag/blob/31c664cc2f5fa0d4b81067a9e64ecf0003e3c16e/adafruit_magtag/peripherals.py#L86](https://github.com/adafruit/Adafruit_CircuitPython_MagTag/blob/31c664cc2f5fa0d4b81067a9e64ecf0003e3c16e/adafruit_magtag/peripherals.py#L86)

makermelissa, 2020-12-03:
I got the numbers from some code I got from @ladyada, but hadn't done any verifications. So it's possible the numbers may need to be adjusted in the MagTag library.

dhalbert, 2020-12-03:
My impression was that the ADC might not be linear, and that the range might be limited, but I haven't started testing yet.

tannewt, 2020-12-03:
@hierophect Will have context on this.

anecdata, 2020-12-03:
Should be linear enough between 150 ~ 2600 mV, which covers the range for 50% of LiPo voltage. AnalogIn does apply  calibration based on device-specific data in efuse.

But whereas [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/adc.html) say we should be able to measure voltages from 0-3.3v at 11dB attenuation, I can't get quite to zero, or above about 2.7v, with the current build.

ladyada, 2020-12-03:
we divide by 2 fyi - so you should have full range

hierophect, 2020-12-04:
What specifically is wrong with the voltage in testing? There are a number of caveats to the S2 ADC readings.

The relevant section to the voltage limits problem is [here](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32s2/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t). When I asked Espressif about this issue, they confirmed that their voltage readings capped out at 2.6V. The section also notes:

> At 11 dB attenuation the maximum voltage is limited by VDDA, not the full scale voltage.

So I guess this could vary depending on whether the VDDA implemented at the module or client PCB level. @anecdata I don't see anywhere in the docs that indicates it the ESP32S2 will measure anything below 100mV, that's been an issue on the ESP32 as well, and it's noted in the above function docs.

The ESP_ADC_CAL component is supposed to fix some of this, but the implementation for the S2 is less fully implemented than for the ESP32. I've looked at the latest master for the [esp_adc_cal](https://github.com/espressif/esp-idf/blob/master/components/esp_adc_cal/esp_adc_cal_esp32s2.c) component and they haven't added any additional features, so we aren't missing anything from a recent update either.

hierophect, 2020-12-04:
You can see some examples of the voltage curves on the ESP32 [here](https://esp32.com/viewtopic.php?f=12&t=1045). The ESP32 IDF implementation actually has the LUT lookup suggested by posters in that thread, that maps ADC readings to better values for nonlinear regions (though I'm not sure how much that can fix the low-end limit). But the ESP32 S2 doesn't have this implemented yet.

anecdata, 2020-12-04:
I can measure 0.000v-3.299v (calibrated) on ADC1 on the ESP32 (using CircuitPython!), so I think Espressif has solved it there.

But for 1/2 of the battery voltage, we are well within the recommended linear range on the ESP32-S2.

I may be missing something, but is the library calculation correct (see my first comment)? `analogin` [returns](https://github.com/adafruit/circuitpython/blob/fb90f51082ab163cd3ee2572aad01fcf0063de90/ports/esp32s2/common-hal/analogio/AnalogIn.c#L93) 0-65535 value for voltage relative to 3.3v if I'm reading it right, so the library should multiply by 3.3 (rather than 2.6) to get back to a voltage?

hierophect, 2020-12-04:
> ADC1 on the ESP32 (using CircuitPython!)
???

I see what you mean in your comment now, though. Yes, I had it multiply by 3.3 to align with the ADC readings of other Circuitpython boards and the documentation, even if it can't get all the way up to the max at whatever VDDA we're supplying. So you're right, that later adjustment should be 3.3, not 2.6.

ladyada, 2020-12-04:
@dhalbert i havent tried this for a while, please see if you get good numbers now

anecdata, 2020-12-04:
@hierophect We have analogin in ESP32SPI-NINA :-)

dhalbert, 2020-12-04:
@ladyada will do, and I'll also see about adding the calibration code to the -S2 implementation.

hierophect, 2020-12-04:
@dhalbert we already use the calibration code that's implemented in the IDF, which is just the basic calibration curve without an LUT. If we want our own LUT, we'll need to decide how to generate it.

dhalbert, 2020-12-04:
I changed the multiplier, and the battery voltage is closer, but is reading about 0.15v low. I tried at 4.2, 3.7, and 3.35v, which are the variable values available from the Monsoon power monitor. I double-checked that the Monsoon is accurate with an external voltmeter.

I have not looked at the code yet.

hierophect, 2020-12-04:
It's suspicious to me that the voltage curve starts at 150mV and is also off by that amount. Maybe the calibration curve offset isn't being applied properly?

dhalbert, 2020-12-04:
I did some instrumentation of `esp_adc_cal_esp32s2.c`:

- The `low_curve` and `high_curve` tables are not used. In fact the comment makes it clear:
```c
    // these values are not used as the corresponding calibration themes are deprecated.
```
- There _are_ efuse values for `adc_calib_low` and `adc_calib_high`. These values are non-zero, at least on my ESP32-S2. So the code does a `characterize_using_two_point()` with those calibration values. Nevertheless, the resulting voltage values are off by about 150mV as noted.

I haven't read all the support issues and surrounding doc yet.

dhalbert, 2020-12-05:
On a Metro ESP32-S2, I fed its DAC to the ADC. I measured the voltage with the ADC, and also with a good multimeter and an ADS1115. I don't see a 150mV difference. The readers differ only by only a few mV, often only 3 or so. I did a similar test with a 1.5V battery on a MagTag, reading its voltage on D10 and also on a multimeter, with a similar minimal difference, certainly not 150mV

I'm thinking any issues with the voltage reading may have to do with the input impedance of the divider. I thought the capacitor would make up for that, but maybe not.

dhalbert, 2020-12-07:
> I changed the multiplier, and the battery voltage is closer, but is reading about 0.15v low. I tried at 4.2, 3.7, and 3.35v, which are the variable values available from the Monsoon power monitor. I double-checked that the Monsoon is accurate with an external voltmeter.

I repeated these measurements _in situ_ with a battery, and I don't see this 150mV offset. I'm not quite sure why I saw it the first time with the Monsoon, but I don't see any offset with the battery and a voltmeter, and the corrected library (https://github.com/adafruit/Adafruit_CircuitPython_MagTag/pull/33), which now uses the proper 3.3V multipler. So I'm going to close this for now.

closed 2020-12-07 by dhalbert

ladyada, 2020-12-07:
hmm ok!

kevinjwalters, 2021-01-10:
I just noticed this too while testing an Enviro+ FeatherWing. The voltage is read from its 3 gas sensors with analogue inputs and the value differed to that seen on a Feather nRF52840 Express.

dhalbert, 2021-01-10:
@kevinjwalters If you are seeing differing measurements below the 2.63V limit compared with another board, go ahead and open an issue with some example code and measurements. Thanks.

kevinjwalters, 2021-01-10:
@dhalbert I've left the FeatherS2 side by side with an nRF52840. I may have a more accurate ADC somewhere, I'll rummage.

Is the ESP32-S2 behaving according to its documentation? It sounds like with -11dB to input signal it should be viable to read up to 3.3V but that's clearly not the case at the moment. Do Expressif have any sample code for reading 3.3V, I've not found any yet.

@anecdata Are you saying the ESP32 and ESP32-S2 differ here?

kevinjwalters, 2021-01-10:
I think I've picked two inputs on the pair of ADCs and curiously there are different ceiling values coming out of these:

```
>>> a1v = a1.value ; d5a10v = d5a10.value ; print(a1v, bin(a1v)) ; print(d5a10v, bin(d5a10v))
52149 0b1100101110110101
51574 0b1100100101110110
```

BTW, is there a reason why the width is not set for ADC2?

https://github.com/adafruit/circuitpython/blob/f15e2d86c0fb3235a9934b74d22a0d70ea2b4f4c/ports/esp32s2/common-hal/analogio/AnalogIn.c#L66-L70

anecdata, 2021-01-10:
ESP32-S2 width is enforced to 13 bits. As far as I can tell, latest Espressif docs indicate ADC should be able to handle up to 3.3v (linearity may well suffer above 2.6v, even with the calibration). We either need to demonstrate 3.3v in another IDE (or demonstrate that it is similarly limited, and file an issue with Espressif), or ask someone at Espressif.

dglaude, 2021-01-10:
Wrote a little piece of code that read on A0 the voltage written on A1:
```
import board
import time
from analogio import AnalogOut, AnalogIn
analog_in = AnalogIn(board.A1)
analog_out = AnalogOut(board.A0)
while True:
    for i in range(0, 65535, 512):
        write = i
        analog_out.value = write
        time.sleep(0.02)
        read=analog_in.value
        print((write, read))
```

It show something linear until it reach a plateau:
> (0, 1886)
> (512, 2124)
> (1024, 2561)
...
> (54784, 53758)
> (55296, 54056)
> (55808, 54413)
> (56320, 54413)
...
> (64512, 54413)
> (65024, 54413)

kevinjwalters, 2021-01-10:
@dglaude Interesting, your FeatherS2 can "reach" higher than mine:

```
(0, 2303)
(512, 2621)
(1024, 3038)
...
(53248, 51534)
(53760, 52010)
(54272, 52149)
(54784, 52149)
...
(64512, 52149)
(65024, 52149)
```

hierophect, 2021-01-10:
I'm not sure what the new issue is here - I confirmed with Espressif that the S2 cannot achieve readings above around 2.6V, so I think that's that. The ESP32 and ESP32S2 do differ in this regard, the original ESP32 has a more sophisticated calibration process including a lookup table (LUT) and the ESP32S2 does not. You could use a voltage divider and diode offset to put your measured voltages in a better range, if necessary.

anecdata, 2021-01-10:
I think the issue is that the Espressif docs are either: wrong, or; anticipating further development to get the ESP32-S2 to comparable ADC behavior as the ESP32. Espressif should go on record with an answer.

hierophect, 2021-01-10:
@anecdata where in the docs do you see the claim that Analog readings should be able to go to 3.3V? I'm going over the page and they say that it's 2.6V everywhere that I can find https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/adc.html

anecdata, 2021-01-10:
[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t)

> When the analog voltage supply (VDDA) is 3.3 V:
> ...
> 11 dB attenuation (ADC_ATTEN_DB_11) gives full-scale voltage 3.9 V (see note below)
> ...
> Note
> At 11 dB attenuation the maximum voltage is limited by VDDA, not the full scale voltage.

hierophect, 2021-01-10:
Yeah I had also just landed on that. I think that this also doesn't imply that 3.3V should be possible, it's just written in a confusing way. I think it means that the maximum value of the integer value would correspond to 3.9V, as in the integer max, but it will platau before ever actually hitting that value. Correspondingly, the maximum voltage is _limited_ by VDDA, but that doesn't necessarily mean it _is_ VDDA. Definitely some word salad going on there.

The header functions above it, however, all refer to a max reading of 2.6V:
> ADC_ATTEN_DB_11 = 3 
> The input voltage of ADC will be attenuated, extending the range of measurement to up to approx. 2600 mV.

dhalbert, 2021-01-10:
> [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t)

Note that's the ESP32, not ESP32-S2 docs.

hierophect, 2021-01-10:
@dhalbert it's actually pretty obviously copy-pasted between the two, the ESP32S2 one is the same:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t

Again, I think they did that because it doesn't _technically_ say what the actual maximums are. But this whole thing could absolutely be a lot more clear. They should really specify outright what the maximum voltages are at 3.3V VDDA and what linearity issues they have / how those are compensated for in the calibration process.

anecdata, 2021-01-10:
sorry, wrong link, esp32-s2 reads the same [https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t)

dhalbert, 2021-01-10:
Ah, sorry, I should have checked the other doc.

anecdata, 2021-01-10:
The odd thing to me, helped along by confusing docs, is that people coming from ESP32 could read the docs and expect 3.3v just like ESP32.

hierophect, 2021-01-10:
@anecdata I agree. ESP32 says that the maximum "recommended" voltage is ~2.6mV, but can go above that, with caveats. ESP32S2 says the same thing but apparently cannot go above it *at all*.

kevinjwalters, 2021-01-10:
@hierophect Is the statement from Espressif available somewhere for viewing? It seems odd they cripple the ESP32-S2 and make it incompatible with ESP32, did they say why they did this? And why are they referring to it as -11dB (i.e. 3.90V -> 1.10V) if it's not?

hierophect, 2021-01-10:
Perhaps that with a higher VDDA value than 3.3V, a higher voltage would be possible? But I don't know if that's even allowed by the voltage specifications.

anecdata, 2021-01-10:
It is a documentation issue, at least for now. ESP32 docs read the same, but with VDDA at 3,.3v, you can still get 3.3v ADC read on ESP32. If this is permanent, the fix is easy for them. If it's a future feature to support full-range, you'd think they could indicate that.

hierophect, 2021-01-10:
@kevinjwalters It was an email chain, and I would hesitate to call it a statement, just that the engineer on the other end confirmed that the maximum value they could obtain from the device was 2.6V. It was an old ticket, from back last fall when I was implementing this.

Again, I don't think they're actually doing anything duplicitous here. The header functions do state that the maximum attainable voltage is 2.6V, and the 11dB measurement *is* on a scale up to 3.9V, it just doesn't actually hit every value in that range.

So, on the ESP32, you can actually hit a full 3.3V reliably, using these same functions? I just checked out the ESP32 page, and it also claims to cap out at 2.6V. If you're obtaining higher voltages than that, it might actually be considered to be undocumented/unofficial functionality.

hierophect, 2021-01-10:
But in either case, there's a clear discrepancy here. I'll open a new ticket with them and see if I can clear any of this up.

anecdata, 2021-01-10:
@hierophect just for completeness on your ESP32 question... The documentation has evolved a bit over time in how they describe analog range in various sections, and looking at code history it looks like calibration functions evolved as well. I ran the same ADC1 example as above on a Feather ESP32 with current IDF, and it has a range of 142mV-3124mV (calibrated voltage from `esp_adc_cal_raw_to_voltage()`; raw reading at the high end is 4095).

Hopefully Espressif can confirm the supported ADC voltage range (linearity aside), and align docs with that.

kevinjwalters, 2021-01-10:
@anecdata Does it plateau at top end or is 3124mV actually equivalent to 3.300V?

anecdata, 2021-01-10:
@kevinjwalters Calibrated ADC read tops out at 3124mV on that ESP32 with current ESP-IDF. That corresponds to raw 12-bit reading of 4095. Meter-measured voltage is 3.3v.

kevinjwalters, 2021-01-10:
@anecdata But does metered 3.20V and 3.25V also correspond to 3124mV?

anecdata, 2021-01-11:
Yes, once the raw ADC bits are maxxed out, the calibrated voltage gets no higher (e.g., 3124mV), regardless of increased voltage applied and read on meter.

In other words, meter reads near 3124mV when raw ADC is at 4095 (and calibrated voltage is at 3124mV) - meter then measures higher as voltage is increased, but ADC bits and calibrated voltage read are already maxxed out and don't change.

hierophect, 2021-01-11:
@anecdata - I still don't understand, you're saying that the ADC actually reaches the maximum value of the bit range (0xFFF, 12 1s in binary) and it corresponds to 3124mV? I'm extremely confused by that, the full scale voltage is supposed to be 3.9V, and my raw returns never approach the full scale value, that's what I've meant by referring to the plateau. I don't see how that would correlate to the documentation in any of the interpretations we've discussed here. Can you elaborate?

anecdata, 2021-01-11:
What I see, using the Espressif ADC example, on ESP32 and ESP32-S2, is that the 12- or 13-bit ADC reading (64x multisampled) hits the maximum value (4095 or 8191) at some voltage well under 3.3V. Similarly, the calibrated value from `esp_adc_cal_raw_to_voltage()` also hits its maximum value at that point. At that point, calibrated value roughly equals the meter-measured voltage. But, increasing the applied voltage cannot increase the raw value because all bits are `1`, and it does not increase the reported calibrated value.

I think the 3.9V is a red herring and just a mathematical artifact. The datasheet says max VDDA is 3.6V, and in reality for us at least it's 3.3V.

hierophect, 2021-01-11:
Hmm. Your observation is backed up by this forum post: https://www.esp32.com/viewtopic.php?f=12&t=1045. I don't understand why they even mention 3.9V in that case, it seems entirely irrelevant if this is the actual voltage curve?

Anyway I'll write up my ticket to them today.

dhalbert, 2021-01-11:
Maybe y'all are talking about measurements done with different attenuation values?

hierophect, 2021-01-11:
@dhalbert while I thought that might be it, if the forum post is to be believed (and it does seem to match @anecdata's experience), this is indeed happening at 11dB. I've been operating under a model I had in my head from reading the documentation and working with the S2 but I clearly need to actually sit down with my hardware and do more comparisons myself.

anecdata, 2021-01-11:
Everything I've said refers to 11dB attenuation, in order to try to get the maximum voltage at the high end. Perhaps if VDDA was 3.6V, the chip would deliver higher calibrated voltage when the raw reading was at maximum (4095/8191). But everyone uses 3.3V.

anecdata, 2021-01-11:
Espressif _may_ be able to improve the calibration over time to raise the maximum detectable voltage (the more mature ESP32 goes higher than the -S2, but there could well be chip differences that account for that).

But for now we should probably encourage ESP32-S2 users to use a well-matched voltage divider if they want to measure voltages higher than 2.6V.

kevinjwalters, 2021-01-12:
I followed @dglaude's lead and used the DAC (on SAMD21) to test the four feather boards I own: [Adafruit Forums: Feather ADC comparison including 2.6V limited ESP32-S2](https://forums.adafruit.com/viewtopic.php?f=60&t=174020) - the png is high resolution to allow a close look at the data.

hierophect, 2021-01-13:
@kevinjwalters thanks for posting that! I've included it in my ticket to Espressif.

dhalbert, 2021-01-13:
Reposting image here for reference:

Here's a graphical comparison of the ADCs on four Feather boards including the FeatherS2 which uses the ESP32-S2 which appears to have an unfortunate limitation with its ADC voltage. The Feather M0 Bluefruit DAC is providing the voltage into a 0.1uF capacitor. The boards are:

> FeatherS2 from unexpected maker - red
> Feather nRF52840 Express - purple
> Feather M0 Bluefruit - light brown
> Feather M4 Express - sky blue

[image]

hierophect, 2021-01-13:
It would be nice if we could add an original ESP32 to this graph for comparison

kevinjwalters, 2021-01-13:
Sadly I don't have an ESP32 but code is easy to run and is over at https://github.com/kevinjwalters/circuitpython-examples/tree/master/adc/adc-comparison. There's a collection program in there too which does the time stamping. I ran that on a desktop but shut down all other applications to minimise load to try and keep the timestamping accurate.

[5 more replies]