adafruit/circuitpython #3788: frequencyio.FrequencyIn  crash on Metro Grand Central

cls22, 2020-12-03:
On the Metro Grand Central, the frequencyio.FrequencyIn functionality seems to be crashing the system:
Adafruit CircuitPython 6.0.0 on 2020-11-16; Adafruit Grand Central M4 Express with samd51p20
>>> import board
>>> import frequencyio
>>> frequency = frequencyio.FrequencyIn(board.A3)
>>>
At this point the unit re-enumerates on USB:-
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
MicroPython NLR jump failed. Likely memory corruption.
Press any key to enter the REPL. Use CTRL-D to reload.

I have tried on a couple of pins with the same result.
Currently there is nothing connected to the board to drive a signal.

dhalbert, 2020-12-19:
I also got this to crash on a Metro M4 with board.A3 and board.D0. The pin does not seem to matter much.
I single-stepped and it seems to be crashing in `tc_reset()`. The argument looked OK: it was the right address for `TC7` when I was debugging on the Grand Central.

dhalbert, 2020-12-21:
This also crashes on a Metro M0, so it's not restricted to SAMD51. I made a build with `frequencyio` enabled.

dhalbert, 2020-12-22:
The problem is that we are hanging indeifinitely in the `while` loop in the code below. 
```c
void tc_reset(Tc* tc) {
    tc->COUNT16.CTRLA.bit.SWRST = 1;
    while (tc->COUNT16.CTRLA.bit.SWRST == 1) {
    }
}
```

`COUNT16.CTRLA.bit.SWRST` is actually a write-only bit. I believe we are checking for SWRST finished in an incorrect wayI think it should be:
```c
    while (tc->COUNT16.SYNCBUSY.bit.SWRST != 0) {
```
We need to check the `SYNCBUSY` register bit, not the bit we set. There are other places where we should be checking but aren't, mostly in `PWMOut.c`. Also there are some other non-`SWRST` settings that should similarly be checking `SYNCBUSY` but are checking the bit that was set instead.it

I started preparing a `samd-peripherals` PR and also a `circuitpython` PR. **However**, making this change does not fix the problem in 6.0.0, and does not explain why it used to "work". So there's still something else wrong.

I need to work an another bug that is higher priority and will come back to this later.

dhalbert, 2020-12-22:
After tracing this more carefully this is due to raising an exception in an interrupt handler, which does not work properly:
https://github.com/adafruit/circuitpython/blob/8b03951fee166b1eea06d114fc347982b7b186bf/ports/atmel-samd/common-hal/frequencyio/FrequencyIn.c#L85.
This is inside `frequencyin_emergency_cancel_capture()`, which is called from `frequencyin_interrupt_handler()`.

`frequencyin_emergency_cancel_capture()` gets called almost immediately after the `FrequencyIn()` is created. If I use the REPL test in the original post, I get the `>>>` after the line that creates, and then it crashes.

This error was not raised in 5.3.1, but is raised in 6.0.0, perhaps due to some difference in ticks and time-keeping

Simply commenting out the `mp_raise_RuntimeError()` prevents the crash. The `FrequencyIn` object is still functional: I tested it with a `PWMOut()` running at 500Hz and got the expected value.

A quick fix is just to comment out the `mp_raise`. That is what we can do to fix the error for 6.0.1. @tannewt, is this OK by you? In the long run, there needs to be some other way for `FrequencyIn()` to signal it is in trouble, and perhaps it should not be in trouble here. There could be an attribute that indicates an error, or, since it pauses itself, there could be a `.paused` attribute the code could check.

Pinging @sommersoft for any comment.

tannewt, 2020-12-23:
@dhalbert Yup, fine by me.

PR #3867 (2020-12-23): FrequencyIn: do not raise in interrupt handler
Fixes #3788.

More info in https://github.com/adafruit/circuitpython/issues/3788#issuecomment-749839525:

Remove this `mp_raise`:
https://github.com/adafruit/circuitpython/blob/8b03951fee166b1eea06d114fc347982b7b186bf/ports/atmel-samd/common-hal/frequencyio/FrequencyIn.c#L85.
This is inside `frequencyin_emergency_cancel_capture()`, which is called from `frequencyin_interrupt_handler()`. Interrupt handlers should not raise exceptions.

Simply commenting out the `mp_raise_RuntimeError()` prevents the crash. The `FrequencyIn` object is still function [... cut]
closed 2020-12-23 by tannewt