adafruit/circuitpython #3789: Add access to backup RAM to save state during deep sleep

dhalbert, 2020-12-03:
Most chips with deep sleep capability provide backup RAM that remains energized during deep sleep. Allow access to this RAM, say via a `microcontroller.nvm`-style mechanism. This will allow a deep-sleeping program to remember state without using flash.

tannewt, 2020-12-03:
Note that there are also retained registers usually too. The S2 has 8 32 bit retained registers in the RTC. We'll either want multiple memory segments or one oddly sized memory where you fill it from the beginning and the retention adapts accordingly. This adaptation can also match RAMs that can have varied retention sizes.

dhalbert, 2021-06-15:
Done as part of sleep implementation.

closed 2021-06-15 by dhalbert