adafruit/circuitpython #3793: Magtag does not boot to CP after flash

jerryneedell, 2020-12-04:
FYI - I just updated to the current main CP repo -- now at 6.1.0-beta2 -- when I build and flash my magtag -- it will only boot to the MAGTAGBOOT bootloader after powercycle or reset -- then I have to copy over the firmware.uf2 file manually -- it runs fine but that is a new step -- has something changed. Are others seeing the same?
here is the sequence I am using 
```
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ make BOARD=adafruit_magtag_2.9_grayscale flash PORT=/dev/ttyACM0
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
ninja: Entering directory `build-adafruit_magtag_2.9_grayscale/esp-idf'
[1/1] Performing build step for 'bootloader'
ninja: no work to do.
esptool.py --chip esp32s2 -p /dev/ttyACM0 --no-stub -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x0000 build-adafruit_magtag_2.9_grayscale/firmware.bin
esptool.py v3.0-dev
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:06:8e:a8
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Erasing flash...
Took 7.92s to erase flash block
Wrote 1302528 bytes at 0x00000000 in 29.6 seconds (351.5 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/esp32s2$ cp build-adafruit_magtag_2.9_grayscale/firmware.uf2 /media/jerryneedell/MAGTAGBOOT/
```

It works fine after loading the .uf2

On my metro_esp32s2  -- it boots to CIRCUITPY normally after flashing.

dhalbert, 2020-12-04:
I do not see this: I do not have the UF2 bootloader on my MagTag and have been loading the `.bin` firmware files via esptool.py. I just tried the 6.1.0-beta.2 download and it's the same. I think it has to do with whether you have the bootloader on your board or not.

jerryneedell, 2020-12-04:
odd -- I don't appear to have a UF2 Bootloader on the MAgTag -- at least it does not got to the bootloader on double tap of reset.   I am just following the same practice I always have but something has changed.  Not a big deal but different.

dhalbert, 2020-12-04:
Could you try backing up your CIRCUITPY and then erasing the MagTag flash completely with esptool.py `erase_flash`? Then try the plain .bin again.

jerryneedell, 2020-12-04:
@dhalbert OK -- That fixed it. Thanks

dhalbert, 2020-12-04:
Then you could try loading the UF2 bootloader again and see its behavior.

jerryneedell, 2020-12-04:
OK -- I'm fine staying with the DFU bootloader but apparently I had the UF2 bootloader but it clearly was not working properly since I could not get to it. But that may be because I used esptool to load the bin and corrupted it....

jerryneedell, 2020-12-04:
I reloaded the uf2 bootloader and it works -- just have to work on the timing of the resets!! But id I then use the esptool to refresh via DFU bootloader, it will not come out of the bootloader -- I have to manually copy the CP .uf2 to it...
reloading the UF2 Bootloader -- just have to remember not to flash via esptool!

jerryneedell, 2020-12-04:
Should we close this? Is there an issue to be fixed?

jerryneedell, 2020-12-04:
I guess the lesson is that the UF2 bootloader and esptool are mutually exclusive.

dhalbert, 2020-12-04:
I think it might be worth discussing in an issue in the bootloader repo, perhaps it's a documentation question.

tannewt, 2020-12-04:
@jerryneedell do `flash-circuitpython-only`. The normal flash target will overwrite the second stage bootloader that does double tap detection and also overwrite OTA data which track what to run next. It defaults to the `factory` partition which is where UF2 lives.

jerryneedell, 2020-12-04:
I should have read the makefile...

tannewt, 2020-12-04:
No problem! It's good to explain why it works the way it does. More and more boards have multiple boot stages.

jerryneedell, 2020-12-26:
Should this be closed?

tannewt, 2020-12-28:
Yup! I think so.

closed 2020-12-28 by tannewt