adafruit/circuitpython #3794: Support for MX25L12833F flash device on QT Py

tkircher, 2020-12-05:
I recently got a QT py and installed a Macronix MX25L12833F 16 MiB flash chip. I noticed it wasn't supported by default, but after adding an entry (see below) in `supervisor/shared/external_flash/devices.h` and then setting `EXTERNAL_FLASH_DEVICES = MX25L12833F` in `ports/atmel-samd/boards/qtpy_m0_haxpress/mpconfigboard.mk`, the new build worked fine.

Maybe not an issue, exactly, but it would be nice to have a way to specify flash devices on the command line for boards like the QT Py where they aren't already populated. Also someone please add this to the repo. ;)

```
// Settings for the Macronix MX25L12833F 16MiB SPI flash
// Datasheet: https://www.macronix.com/Lists/Datasheet/Attachments/7447/MX25L12833F,%203V,%20128Mb,%20v1.0.pdf

#define MX25L12833F {\
    .total_size = (1UL << 24), /* 16 MiB */ \
    .start_up_time_us = 5000, \
    .manufacturer_id = 0xc2, \
    .memory_type = 0x20, \
    .capacity = 0x18, \
    .max_clock_speed_mhz = 133, \
    .quad_enable_bit_mask = 0x40, \
    .has_sector_protection = true, \
    .supports_fast_read = true, \
    .supports_qspi = true, \
    .supports_qspi_writes = true, \
    .write_status_register_split = false, \
    .single_status_byte = true, \
  }
```

tannewt, 2020-12-07:
Hi! Please create a PR to add the flash definition. Instructions are here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github Thanks!

skerr92, 2020-12-14:
Was this SPI Flash added? If not I can added it to my open PR for adding the AT25DF641A SPI Flash.

tkircher, 2020-12-15:
> Was this SPI Flash added? If not I can added it to my open PR for adding the AT25DF641A SPI Flash. #3824 See this PR for the issue #3782.

Haven’t submitted a PR for this. Please go ahead and add it!

skerr92, 2020-12-16:
Added this in PR #3824, can be closed when PR is merged.

dhalbert, 2020-12-21:
Fixed by #3824.

closed 2020-12-21 by dhalbert