adafruit/circuitpython #3795: CRC failure reading word in 6.1.0-beta.2

FoamyGuy, 2020-12-05:
This basic code for the battery fuel gauge succeeds on beta.1 but fails with this error on beta.2:
```
>>> import board
>>> from adafruit_lc709203f import LC709023F
>>> sensor = LC709023F(board.I2C())
>>> print("IC version:", hex(sensor.ic_version))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 124, in ic_version
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
```
I would guess it's related to bus device moving into the core. But I'm not really sure what the specific error means.

ladyada, 2020-12-05:
probably isnt handling clock stretching properly

tannewt, 2020-12-07:
@FoamyGuy what board are you on?

FoamyGuy, 2020-12-07:
@tannewt It was definitely ESP32-S2 port. I tested some on MagTag and some on Unexpected maker Feather S2. I'm not 100% sure if I tested both versions on both devices or not though.

dhalbert, 2020-12-08:
We should test on nRF and SAMD also. Unfortunately I don't have this sensor.

I did stumble across this interesting post about adding clock stretching to a different fuel gauge sensor, while I was looking for something else: https://www.esp32.com/viewtopic.php?f=13&t=15999

FoamyGuy, 2020-12-08:
I can try out out on those ports sometime today. I'll make note of my findings here.

FoamyGuy, 2020-12-08:
On: `Adafruit CircuitPython 6.1.0-beta.1 on 2020-11-20; Adafruit Hallowing M4 Express with samd51j19`
```
>>> sensor = LC709023F(board.I2C())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 102, in __init__
  File "adafruit_lc709203f.py", line 135, in power_mode
  File "adafruit_lc709203f.py", line 195, in _write_word
  File "adafruit_lc709203f.py", line 195, in _write_word
  File "adafruit_bus_device/i2c_device.py", line 104, in write
TypeError: extra keyword arguments given
```
Results seem to be the same with and without bus_device in my lib folder.

On beta.2 it's back to the same as above:
```
Adafruit CircuitPython 6.1.0-beta.2 on 2020-12-03; Adafruit Hallowing M4 Express with samd51j19
>>> import board
>>> from adafruit_lc709203f import LC709023F
>>> sensor = LC709023F(board.I2C())
>>> print("IC version:", hex(sensor.ic_version))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 124, in ic_version
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
```

dhalbert, 2020-12-08:
I'm confused about why this is happening, based on reading the code. I thought perhaps the new native implementation was faster. You could try adding some short delays in various places in the driver .py file. I wish I had one of these to test, or something that exhibits similar behavior.

FoamyGuy, 2020-12-09:
Strangely on nRF the beta.1 has the CRC error as well:
```
Adafruit CircuitPython 6.1.0-beta.1 on 2020-11-20; Adafruit Feather Bluefruit Sense with nRF52840
>>> import board
>>> from adafruit_lc709203f import LC709023F
>>> sensor = LC709023F(board.I2C())
>>> print("IC version:", hex(sensor.ic_version))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 124, in ic_version
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
```

same on beta.2
```
Adafruit CircuitPython 6.1.0-beta.2 on 2020-12-03; Adafruit Feather Bluefruit Sense with nRF52840
>>> import board
>>> from adafruit_lc709203f import LC709023F
>>> sensor = LC709023F(board.I2C())
>>> print("IC version:", hex(sensor.ic_version))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 124, in ic_version
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
```
adafruit_bus_device and adafruit_lc709203f libraries tested were from 20201208 bundle.

jepler, 2020-12-18:
Here's what an attempt to read cell voltage looks like on esp32s2 near the tip of main (f2204d7d88b505155cd25b38b71ae3c35b86d438):

[image]

jepler, 2020-12-18:
Somehow, this _IS_ due to `adafruit_bus_device`-in-core.  I substituted the old Python-coded one by trickery.  I copied it to a folder `old_bus_device` so that I could use the following sequence in the repl:
```
>>> import old_bus_device
>>> import sys
>>> sys.modules['adafruit_bus_device'] = old_bus_device
>>> import adafruit_bus_device
>>> adafruit_bus_device
<module 'old_bus_device' from '/lib/old_bus_device/__init__.py'>
>>> import 
>>> import board, busio, adafruit_lc709203f; bus = busio.I2C(board.IO10, board.IO11); dev = adafruit_lc709203f.LC709023F(bus); print(dev.cell_voltage)
3.807
```

Here's the new transaction -- first the zoomed in read portion, then the whole zoomed out  transaction:
[image]
[image]

The difference seems to be that with the built-in `adafruit_bus_device`, there is a STOP condition generated after the address is sent, at about 250us.

ladyada, 2020-12-18:
yeah we *default* to repeated start when using write_then_readinto...because linux does and cannot otherwise. if you need to generate a stop (which some chips need) you have to do a separate write and then a separate readinto

closed 2020-12-18 by jepler

jryan230, 2021-05-14:
Hi, I'm having this same error with this fuel gauge. I'm not a developer so I can't quite tackle this, but I have the Feather nRF52840 Express and this fuel gauge. I'n getting this same error: 

Traceback (most recent call last):
  File "code.py", line 454, in <module>
  File "adafruit_lc709203f.py", line 122, in cell_percent
  File "adafruit_lc709203f.py", line 188, in _read_word
RuntimeError: CRC failure on reading word

when I try and use this sensor with the code:

---------
import time
import board
from adafruit_lc709203f import LC709203
 
sensor = LC709203F(board.I2C())

while True: 

    print(str(sensor.cell_percent))
-------

e-simpson, 2021-06-11:
I'm also having this issue on:

- latest 6.3.0 CircuitPython & libraries
- adafruit clue board (nrf52840)

PR #3847 (2020-12-18): adafruit_bus_device: Don't transmit STOP condition in write_then_readinto
Testing performed: on esp32s2 kaluga with LC709203 sensor on the pins shown below, use the repl to check the cell voltage:
```
>>> import board, busio, adafruit_lc709203f; bus = busio.I2C(board.IO10, board.IO11); dev = adafruit_lc709203f.LC709023F(bus); print(dev.cell_voltage)
3.807
```

(this bug is not esp32s2 specific and was reported to affect other ports as well, but the LC7… sensor is sensitive to this bug in a way other sensors apparently are not)

Closes #3795
commit 6c4df5a8 (2020-12-18): adafruit_bus_device: Don't transmit STOP condition in write_then_readinto

Closes #3795