adafruit/circuitpython #3798: be able to setup SPI (or any other SERCOM func) from pins A1-A4 MatrixPortal M4

ricardoquesada, 2020-12-07:
MatrixPortal M4 has pins A1-A4 exposed to the user, they are SERCOMs pins[1], but failed when trying to setup SPI with them.

What I did was:
```python
import board
import busio

spi = busio.SPI(board.A2, MOSI=board.A4) #I tried different variations, all failed.
```

And I get:

```
ValueError: Invalid pins
```

Is there a workaround? 
What should I do to use pins A1-A4 for another SPI (using CircuitPython), in the MatrixPortal M4?

Thanks!

[1]: https://cdn-learn.adafruit.com/assets/assets/000/095/095/original/led_matrices_MatrixPortal_M4_Schematic.png?1601064911 )

dhalbert, 2020-12-07:
These four pins supply all the pads on SERCOM0, so it should work, but perhaps another piece of functionality is grabbing SERCOM0, or there's an error in our pin table. We will investigate - thanks.

Test program:
```python
import board
import busio

pins = (board.A1, board.A2, board.A3, board.A4)
for sck in pins:
    for mosi in pins:
        for miso in pins:

            if sck == mosi or sck == miso or mosi == miso:
                continue

            try:
                spi = busio.SPI(sck=sck, mosi=mosi, miso=miso)
            except:
                print("FAILED: sck={}, mosi={}, miso={}".format(sck, mosi, miso))
                continue
            spi.deinit()
            print("Works: sck={}, mosi={}, miso={}".format(sck, mosi, miso))
```

ricardoquesada, 2020-12-07:
Thanks for the script. I was able to find a few valid combinations. This definitely works for me.

Not sure whether all combinations should work though. Adding full output in case this is a bug:
(output contains additional `mp_printf` I added in `SPI.c`)

```
Works: sck=board.A1, mosi=board.A2, miso=board.A3
Works: sck=board.A1, mosi=board.A2, miso=board.A4
pad combination not possible: clk:1, mosi:2, dopo:255
FAILED: sck=board.A1, mosi=board.A3, miso=board.A2
pad combination not possible: clk:1, mosi:2, dopo:255
FAILED: sck=board.A1, mosi=board.A3, miso=board.A4
Works: sck=board.A1, mosi=board.A4, miso=board.A2
Works: sck=board.A1, mosi=board.A4, miso=board.A3
Invalid clock pad: clk:0
FAILED: sck=board.A2, mosi=board.A1, miso=board.A3
Invalid clock pad: clk:0
FAILED: sck=board.A2, mosi=board.A1, miso=board.A4
Invalid clock pad: clk:0
FAILED: sck=board.A2, mosi=board.A3, miso=board.A1
Invalid clock pad: clk:0
FAILED: sck=board.A2, mosi=board.A3, miso=board.A4
Invalid clock pad: clk:0
FAILED: sck=board.A2, mosi=board.A4, miso=board.A1
Invalid clock pad: clk:0
FAILED: sck=board.A2, mosi=board.A4, miso=board.A3
Invalid clock pad: clk:2
FAILED: sck=board.A3, mosi=board.A1, miso=board.A2
Invalid clock pad: clk:2
FAILED: sck=board.A3, mosi=board.A1, miso=board.A4
Invalid clock pad: clk:2
FAILED: sck=board.A3, mosi=board.A2, miso=board.A1
Invalid clock pad: clk:2
FAILED: sck=board.A3, mosi=board.A2, miso=board.A4
Invalid clock pad: clk:2
FAILED: sck=board.A3, mosi=board.A4, miso=board.A1
Invalid clock pad: clk:2
FAILED: sck=board.A3, mosi=board.A4, miso=board.A2
Invalid clock pad: clk:3
FAILED: sck=board.A4, mosi=board.A1, miso=board.A2
Invalid clock pad: clk:3
FAILED: sck=board.A4, mosi=board.A1, miso=board.A3
Invalid clock pad: clk:3
FAILED: sck=board.A4, mosi=board.A2, miso=board.A1
Invalid clock pad: clk:3
FAILED: sck=board.A4, mosi=board.A2, miso=board.A3
Invalid clock pad: clk:3
FAILED: sck=board.A4, mosi=board.A3, miso=board.A1
Invalid clock pad: clk:3
FAILED: sck=board.A4, mosi=board.A3, miso=board.A2
```

had to patch the script.
replace: ` spi = busio.SPI(sck=sck, mosi=mosi, miso=miso)` -> ` spi = busio.SPI(clock=sck, MOSI=mosi, MISO=miso)`

dhalbert, 2020-12-07:
> had to patch the script.
> replace: ` spi = busio.SPI(sck=sck, mosi=mosi, miso=miso)` -> ` spi = busio.SPI(clock=sck, MOSI=mosi, MISO=miso)`

Whoops, that's a great example of why bare `except:` is a bad idea. Glad you got it working.

All combinations definitely don't work: it has to be like this (from the data sheet):
[image]

dhalbert, 2020-12-07:
Closing since there is not actually a bug.

closed 2020-12-07 by dhalbert

ricardoquesada, 2020-12-07:
Thanks for the info!