adafruit/circuitpython #3805: Implementing PA/LNA control in CircuitPython for nrf52840

Embedinno, 2020-12-08:
Hello,

I am working on custom hardware based on nrf52840 IC and running Circuit Python on it.

This hardware have implementation of PA/LNA chip in the antenna section which is controlled via GPIO. Control of the PA via GPIO in user code is working correctly but has a lot of limitations. I am looking into how to implement the automatic PA/LNA switching via soft device which is capable of triggering GPIO pin via PPI features on TX and RX events.. I came across this explanation in Nordic dev. zone:

https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/pa-lna-support-in-s132

Is it possible to access the corresponding settings via circuit python? Or should this function be implemented in custom circuit python build and, if yes, what would be the best way to implement it?

I would appreciate any suggestions.

Thanks,
Dusan

tannewt, 2020-12-08:
Hi Dusan, are these settings you set once and then work automatically? If so, they can go in `board_init`. If not, I'm not sure where the best place would be.

Embedinno, 2020-12-08:
> Hi Dusan, are these settings you set once and then work automatically? If so, they can go in `board_init`. If not, I'm not sure where the best place would be.

Hello, thanks for your reply.

Yes, this is suppose to be set only once for a chosen GPIO pin (which is hw dependent) and after this the pin should be controlled by the soft device whenever there is an TX or RX being done.

I will try to implement definition of this function in board.c file and calling of the same with pin selection within 'board_init' function.

On the other hand, I searched the ports/nrf folder for 'opt.common_opt' (which should contain 'pa_lna') and I found it is used for 'conn_evt_ext' within _bleio/Adapter.c file. Would this be a second place where the mentioned function might be implemented (while still being called from 'board_init')?

All the best,
Dusan

tannewt, 2020-12-09:
I'm not sure. You'll have to see what works.

Embedinno, 2020-12-12:
Hello,

I tried adding the mentioned code (with few little changes) within 'board_init' function, compiled without any errors and tried it with custom hw but its not working.

On the other hand, I tried adding the code in the  _bleio/Adapter.c file within 'ble_stack_enable' function (right after the section where 'opt.common_opt' is used for 'conn_evt_ext'), compiled without any errors and it works!

Question 1: Any idea on the reason why the same code is not working when executed within 'board_init' function? How can I make sure that this function is actually executed during startup?

Question 2: Would it be possible to make this a separate function that I could call within Circuit Python code? Any example how this could be done?

Have a great day,
Dusan

tannewt, 2020-12-15:
> On the other hand, I tried adding the code in the _bleio/Adapter.c file within 'ble_stack_enable' function (right after the section where 'opt.common_opt' is used for 'conn_evt_ext'), compiled without any errors and it works!

Nice!

> Question 1: Any idea on the reason why the same code is not working when executed within 'board_init' function? How can I make sure that this function is actually executed during startup?

It might need to be run after the BLE subsystem is initialized.

> Question 2: Would it be possible to make this a separate function that I could call within Circuit Python code? Any example how this could be done?

It's pretty specific. Do you really want to expose control to Python?

Could you just always enable it? You could add a #define that you can set in the board config which turns it on where you have it working now.

> Have a great day,
> Dusan

dhalbert, 2021-03-31:
@Embedinno How are you doing on this? Would you like to leave this issue open?

Embedinno, 2021-04-05:
Hello,

this issue can be closed. As mentioned in the previous post, I managed to
get the things working but only as fixed and compiled code.