adafruit/circuitpython #3822: OneWireBus does not seem to detect a DS18B20 on featherS2

Neradoc, 2020-12-13:
Tested a DS18B20 temperature sensor (with the appropriate resistor) on a Feather S2 and other boards (like Feather M0 express) and the device is not found on the S2. I think I correctly checked it's connected to the right pin (it throws an error otherwise).

Feather S2:

```python
import board
from adafruit_onewire.bus import OneWireBus
ow_bus = OneWireBus(board.IO17)
devices = ow_bus.scan()
print("Devices:",devices)
```
```
code.py output:
Devices: []
```
Feather M0 Express
```python
import board
from adafruit_onewire.bus import OneWireBus
ow_bus = OneWireBus(board.A0)
devices = ow_bus.scan()
print("Devices:",devices)
```
```
code.py output:
Devices: [<OneWireAddress object at 20001b20>]
```

tannewt, 2020-12-15:
OneWire is bit banged so it might be digitalio or timing related.

dwolbeck, 2020-12-31:
Seems to be the same issue with STM32F405 Cortex M4 CPU on Adafruit Feather STM32F405 Express.
Find my investigations here: [OneWire support STM32F405](https://forums.adafruit.com/viewtopic.php?f=57&t=173451&p=845843&hilit=STM32F405#p845843)

Perhaps stm32duino / Arduino_Core_STM32 group is working on something similar (no. 412)?
Like @tannewt estimated something with timing.

Maybe this post is helpful even it is not CircuitPython related: [Support OneWire device (ex: DS1822) #412](https://github.com/stm32duino/Arduino_Core_STM32/issues/412 )

CrossLAN, 2021-01-05:
For STM32F405 I filed a fresh issue #3935

Hyna, 2021-02-25:
Hello,
I'm having the same issue on ESP32-S2 and nucleoG7. 

with this simple code reset() returns always false:
`import board
from adafruit_onewire.bus import OneWireBus
ow_bus = OneWireBus(board.XXX)
print("Resetting bus...", end="")
if ow_bus.reset():
    print("OK.")
else:
    print("Nothing found on bus.")`

When i probe the bus (on nucleoG7), timing is not correct. According to [source](https://github.com/adafruit/circuitpython/blob/bbe0602738ae5c5d4bfc1cab4a515a82336c396b/shared-module/bitbangio/OneWire.c#L55) there should be common_hal_mcu_delay_us(480); with BUS low. Then bus is switched to input followed by common_hal_mcu_delay_us(70); After that the bus level is read.
In my case, first delay is only about 420 us, which is outside the spec for most of the 1wire devices.
[image] Reset() returns false. 

Even more strange behavior with 1wire eeprom AT21:
[image]

dhalbert, 2022-01-16:
@Hyna see if #5866 helps for you.

dhalbert, 2023-10-28:
Closing in favor of #5865, but will xref to here.

closed 2023-10-28 by dhalbert