adafruit/circuitpython #3825: Long Live Heap feature is worth mentioning in Differences from MicroPython

kbanks-krobotics, 2020-12-15:
The use of both a long and short lifetime heap is different from MicroPython and probably worth mentioning in your docs, for example at

https://circuitpython.readthedocs.io/en/6.0.x/README.html#differences-from-micropython

1. It's an opportunity to brag about a performance optimization :-)
2. It's a pretty significant change behind the scenes, and it might catch people doing VM extensions by surprise

Things that **don't** move around in MicroPython between compile-time and execution-time **can** move around in CircuitPython.

FWIW

kbanks-krobotics, 2020-12-15:
Somebody please add a Documentation label to this - thanks!

tannewt, 2020-12-15:
@kbanks-krobotics Want to make a PR with the addition?

kbanks-krobotics, 2020-12-16:
Insufficient time this evening, but something LIKE:
The CircuitPython VM uses the heap differently than MicroPython, treating the single physical heap as two logical heaps - one for short-lived items, the other for long-lived items. This helps reduce fragmentation, but does mean objects might get relocated between compilation and execution.

tannewt, 2020-12-16:
@kbanks-krobotics how are you hitting this? Are you adding a native module?

kbanks-krobotics, 2020-12-17:
Yes, I was porting over some features from a customized build of MicroPython (on STM32) to CircuitPython (on ESP32S2).
The features are part of a proprietary product line for now, but someday we hope to backport them into their ancestral codebases.
I agree "application-level" programmers are unlikely to be affected by this, but "system-level" programmers...

At the end of the project I will try to do a documentation PR with what I've learned (so far the "fancier heap" is the only architectural level difference I have noticed).

tannewt, 2020-12-22:
Ok, no problem! Please don't advertise it as CircuitPython until the changes are upstreamed.

This only matters when a native module holds onto a pointer usually. The Python references will be updated internally. (We only move at the end of an import before multiple copies can be made.)

tekktrik, 2022-04-28:
@kbanks-krobotics Are you still able to PR this?  I'm doing some documentation things and am happy to PR what you wrote if not!

dhalbert, 2023-10-27:
Rendered moot by #8281, which removed long-lived allocations.

closed 2023-10-27 by dhalbert