adafruit/circuitpython #3826: Low timeout values for BLERadio().start_scan() can lead to safe mode when calling BLERadio().connect()

catequalsgood, 2020-12-15:
I have been using a Feather Sense in a project and noticed that about once a day it would go into safe mode. After some testing I have narrowed it down to specific values for the timeout of `start_scan()`. Large values (for example 0.1) seem fine and the code runs for several days. Very small values (for example 0.01) are simply ignored so that `start_scan()` never times out. Values in between (for example 0.05) run fine for a few hours or sometimes even a day but eventually cause the Feather to go into safe mode. 

My setup for testing was the following: 
One ItsyBitsy nRF52840 running the unmodified ble_adafruit_simpletest.py example code.
One Feather nRF52840 Sense running a modified version of the ble_adafruit_simpletest_client.py example (below). 

```
import time

import adafruit_ble
from adafruit_ble_adafruit.adafruit_service import AdafruitServerAdvertisement
import supervisor

ble = adafruit_ble.BLERadio()

connection = None

print("Scanning for an Adafruit Server advertisement...")
for adv in ble.start_scan(AdafruitServerAdvertisement, timeout=0.05):
    print("about to connect")
    connection = ble.connect(adv)
    print("connected")
    break

ble.stop_scan()

supervisor.reload()

```
The last line printed to the REPL is "about to connect". 
The safe mode message I get is the following:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Nordic Soft Device failure assertion.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

```

For my project I will just increase the timeout to 0.1 but since it took me a while to identify the issue I thought I would share. I did most of the testing running beta versions of 6.0.0 but after updating CircuitPython to the final 6.0.0 and updating the libraries the behavior is still the same.

dhalbert, 2020-12-18:
I believe this may be due to the non-zero timeout being less than the interval, because the default interval is 0.1 seconds. I put in a check for this which will require the timeout either being zero or >= the interval.

dhalbert, 2020-12-30:
Fixed by #3861.

PR #3861 (2020-12-22): BLE fixes
(Same as #3848, but applied against `6.0.x` instead of `main`; MagTag name fix not included because board is not in 6.0.x)

- Characteristic and descriptor `max_length`s can be zero. Fixes #3746.
- Assure that `start_scan()` timeout is either zero or >= interval. This is a guess for this fix, but it coincides with working values reported in #3826. Fixes #3826.
- Assure that `start_scan()` max timeout is not too large for a `uint16_t` when passed to nRF SD. Fixes #3340.
- Fixes small advertising interval values. Fix check and make sure it's at least 0.02001, to allow for rounding issu [... cut]
closed 2020-12-30 by dhalbert