adafruit/circuitpython #3831: the #magtag be listed in the support matrix

d-c-d, 2020-12-16:
see https://circuitpython.readthedocs.io/en/6.0.x/shared-bindings/support_matrix.html

One would expect that the MagTag would in this list, but it is not present today

dhalbert, 2020-12-16:
6.0.x does not support the MagTag: you have to use 6.1.x. The [circuitpython.org downloads page](https://circuitpython.org/board/adafruit_magtag_2.9_grayscale/) for MagTag does not list 6.0.x.

You can find the MagTag listed here: https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html.

However, there is a bug that it's listed just as "MagTag", and not as "Adafruit MagTag". We need to fix that.

d-c-d, 2020-12-16:
Thanks - 
   **is there a link to the latest docs anywhere**?
     (e.g.   https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html )
   When I started at https://circuitpython.readthedocs.io/ I only got to the 6.0.x docs, 
Note that the link from https://learn.adafruit.com/adafruit-magtag?view=all goes to 
   https://circuitpython.readthedocs.io/
   which is loading the 6.0.x docs, not the 'latest'
   
   ( I guess the view all as one page end up bringing in the generic 
      https://learn.adafruit.com/adafruit-magtag/welcome-to-the-community
       link, so that probably explains why ReadTheDocs under magtag still goes to 6.0.x )

tannewt, 2020-12-16:
On Read the Docs you can click the bottom left box to switch versions. It has latest and all of the previous major versions.

PR #3863 (2020-12-22): Add Adafruit to MICROPY_HW_BOARD_NAME for various boards
Fixes #3831.

Add "Adafruit" to  `MICROPY_HW_BOARD_NAME` for a few boards for which it was missing. This makes these boards consistent with the other Adafruit boards. In particular, it makes them easier to find in the Support Matrix in readthedocs.
closed 2020-12-22 by dhalbert