adafruit/circuitpython #3833: storage.remount("/", True) resulting in RuntimeError on 6.1.0-beta.2

kattni, 2020-12-16:
When working through [this CircuitPython storage example](https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage) on QT Py M0, running 6.1.0-beta.2, it fails to work properly - the temperature logging file is never created and the filesystem never switches to read-only. Checking `boot_out.txt`, I find the following traceback:
```
Traceback (most recent call last):
  File "boot.py", line 18, in <module>
RuntimeError: Cannot remount '/' when USB is active.
```
I tried attaching it to a USB battery pack, thinking it meant the USB data connection - this did not resolve the issue, the results are the same. Tried the example on a Trinket running 6.1.0-beta.2, and the results are the same.

I tried the QT Py and Trinket running 6.0.0, and the example works properly.

It appears that something changed between 6.0.0 and 6.1.0, and I am unsure whether it is a bug or a feature.

ladyada, 2020-12-20:
dup of
https://github.com/adafruit/circuitpython/issues/3709
?

kattni, 2020-12-21:
Yes. Moved info there. Closing.

closed 2020-12-21 by kattni