adafruit/circuitpython #3835: LED "glitch" on ESP32-S2

kattni, 2020-12-16:
@hierophect I was asked to tag you on this issue. I am not assigning you as I don't want to make assumptions.

This has been seen on both the MagTag and the Metro ESP32-S2. On the MagTag, I was running 6.1.0-beta.2.

The "glitch" is that while an animation or text is displayed on the LEDs, e.g. they are being lit up, intermittently, a portion of the LEDs will flash a different color.

I am able to consistently reproduce it using the code [here](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/MagTag_Cheerlights_LED_Animations/MagTag_Cheerlights_LED_Animations.py) which requires a MagTag and an LED strip (preferably one of the new JST connector strips) - the strip is plugged into the D10 port on the MagTag, if needed you can check [the guide](https://learn.adafruit.com/cheerlights-led-animations/assembly) for details. When running this code, the last seven or so LEDs on the strip will flash a different color pretty regularly.

Noe and Pedro were seeing the same issue with the code [here](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/IoT_NeoPixel_Sign/code.py) on Metro ESP32-S2 connected to a series of NeoPixel strips for [this project](https://learn.adafruit.com/iot-led-sign/overview). If you watch the GIF that is the main one for the guide (upper left corner of every page), you can see a white LED flash when the "a" and "y" in "Holiday" scrolls past. It's often more pronounced than that, involving more LEDs.

I tried to simplify the code down to something to replicate it without a lot of other variables, but did not succeed. The glitch is not present when only using the LEDs. It can't be the LED Animation library or the MagTag library, because neither are present in Noe and Pedro's project. It is unlikely to be network related because the issue happens on the MagTag between fetches, not during. 

I was however able to get the LEDs to glitch running very simple code when the board soft-resets, on MagTag, with the LED strip mentioned above. Save the following as code.py:
```
import board
import neopixel

pixels = neopixel.NeoPixel(board.D10, 30, brightness=0.3, auto_write=False)

pixels.fill((255, 0, 0))

while True:
    pixels.show()
```
Then modify the file and save it. You should receive a light show. I find manually resetting the board using ctrl+D and ctrl+C doesn't always make it glitch. But the soft reload that comes with writing something to the board does it every time.

I tried to include everything I could think of here, but feel free to ping me if you have any questions.

hierophect, 2020-12-16:
Hmm, probably more RMT troubles. I'll check it out.

djecken, 2020-12-30:
Hi folks, here's a video of the issue in action. Thanks!

https://user-images.githubusercontent.com/869133/103387130-22a7b000-4ad0-11eb-85f6-538351ac3b31.mov

jepler, 2021-01-21:
I can reproduce a problem by running this in the REPL on a Kaluga:
```
>>> b = b'\1\1\1\0\0\0' * 240
>>> d = digitalio.DigitalInOut(board.IO18)
>>> while True: neopixel_write.neopixel_write(d, b)
... 
```
***AND*** continuously writing to the CIRCUITPY drive from Linux:
```
pi@raspberrypi:~/media/CIRCUITPY $ while sync; do touch boot_out.txt ; done
```

Here's a "good" neopixel trace from saleae:
[image]

good trace, zoomed out:
[image]

Here's a "bad" trace, which looks just like the good one to me:
[image]

but what is obviously wrong is that in the "zoomed out" view, it takes MUUUUCH longer:
[image]

Here's how the strip looked with a glitch: (data comes IN from the RIGHT and shifts to the LEFT):
[image]

jepler, 2021-01-22:
I investigated the filesystem access "angle" because I believe I see similar disruptions to the neopixel during UF2 bootloading -- the LEDs are mostly blinking yellow/orange, but occasionally a flicker of another color can be seen.  This potentially points at a problem outside of CircuitPython, or in implementation code that is shared between CP and the UF2 bootloader.  However, Kattni and N+P's scenarios don't seem to involve filesystem/flash access.

kevinjwalters, 2021-01-22:
@jepler Can you get an analogue view of the data? Perhaps the salae isn't showing some strange nastiness in the actual signal due to either thresholding or transient spikes?

andywarburton, 2021-07-13:
Hey folks, I just came across this issue while researching an issue I'm experiencing and it sounds exactly like Kattni described above. I'm using the Tiny S2 from Unexpected maker with a 60 neopixel ring. The animation works just fine apart from random flashes of other colours. It occurs even under the most basic of code conditions. Please let me know if there is anything I can do to help debug.

andywarburton, 2021-07-13:
If it helps, I caught a video of the glitch... it appears towards the end of the video https://www.youtube.com/watch?v=WQKzEdR0QGc

juandoplt, 2021-07-14:
Hi, i have the same issue with a raspberry pico, i use comet animation suddenly some leds show a random color.
I'm using the pl9823 version of neopixels. Any ideas?

hierophect, 2021-07-14:
@juandoplt you might want to make a new issue, if you're experiencing problems with the RP2040. Since the underlying hardware is totally different, it's very unlikely to be the same issue across both vendors.

tannewt, 2021-07-15:
One thing to try would be putting the neopixel code in RAM so that its running doesn't have to contend with flash accesses.

bradanlane, 2022-03-21:
An interesting data point which may (or may not) be of use.

I have a Lolin S2-mini (ESP32-S2) running CircuitPython 7.2.3. The device is driving 240 WS2812C-2020 "neopixels".

It's been glitching just as described above.

On the rare occasion where I plug it into my computer (Windows 10), I might get the "Unrecognized USB device" error. In these cases, the device still receive power but does not have any communications. What is relevant here is that it **does not glitch** in this situation.

jdimpson, 2022-04-13:
So I've been having what might be this problem, on a qt-py-esp32-s2, with 120 neopixels (5x24 circles). Is there a way to tell whether the above RMT driver patch in the micropython code base has made it's way into a circuitpython release?

Also, as yet another data point, the glitch i see matches the other descriptions here (and the video above), however, it only happens when my code does **not** initialize the wifi radio. Specifically, if I comment out this line:
`wifi.radio.connect(secrets["ssid"], secrets["password"])`
the flickering ceases. This is the case regardless of whether the qtpy is connected as a USB disk/serial device or just powered on. Circuit python 7.2.5

crazyiop, 2022-06-01:
using:
Adafruit CircuitPython 7.3.0 on 2022-05-23; S2Mini with ESP32S2-S2FN4R2
Board ID:lolin_s2_mini

I can also confirm this bug and the workaround of @jdimpson .

my project use the neopixel a lot but only rarely get info from the web. In the end, I completely shut down the wifi (`wifi.enabled=False`) between uses to get the flickering stop. This is still not an ideal solution as the whole reconnect, takes some time that is noticeable.

jdimpson, 2022-11-15:
I just saw adafruit is about to offer a NeoPixel BFF for the qt-py form factor. I'm hopeful that as a result, this problem will get more attention. I look forward to synchronizing NeoPixels over wifi!

jepler, 2023-10-30:
might be worth checking if this is any different after the idf upgrade.

crazyiop, 2023-11-04:
@jepler I just updated my board and put an equivalent code (refreshing a lot the pixels while staying connected to the wifi). The issue is still here. 
From memory the glitch is not exactly the same, so I guess the interrupt occurring during the px.show() might have a different timing...

tested on a lolin s2 board, with 8.2.7

nullr0ute, 2023-11-05:
> tested on a lolin s2 board, with 8.2.7

The IDF rebase landed as part of the the 9.x release so it won't be in 8.2.7

crazyiop, 2024-01-11:
I finally retested it with 9.0.0-alpha.6, same board: still got the glitches. (I have not proper metric but it *feel* like there is less of them...)

jepler, 2024-01-11:
I notice that we could use the flag ESP_INTR_FLAG_IRAM to allocate the bounce buffer of the RMT peripheral in internal RAM rather than PSRAM. (the IRQ handler is already in IRAM)

closed 2024-02-15 by dhalbert

crazyiop, 2024-03-06:
I just finally found the time to test again with the latest build available on the s3 (adafruit-circuitpython-lolin_s2_mini-en_US-20240306-main-PR9008-e64082d.bin) and can confirm (if that was needed?) that it works great with my setup that was triggering the issue.

Thanks a lot !!

PR #8920 (2024-02-15): Improve neopixel on ESP
Greedily grab as much RMT memory as we can. It blocks other RMT transmit channels but we only use it temporarily anyway. The more we can grab, the fewer interrupts are needed to keep the transmit going.

Flickers may still happen due to file system writes but most of the time the animation just pauses.

On ESP32, move CircuitPython to the second core. This helps NeoPixel by moving the RMT interrupt to the second core as well.

When testing ESP32 I noticed that settings.toml writes won't apply until after hard reset. This removes that constraint but still requires the pa [... cut]
commit 9537b1dc (2024-02-15): Improve neopixel on ESP

Greedily grab as much RMT memory as we can. It blocks other RMT
transmit channels but we only use it temporarily anyway. The more
we can grab, the fewer interrupts are needed to keep the transmit
going.

Flickers may still happen due to file system writes but most of the
time the animation just pauses.

On ESP32, move CircuitPython to the second core. This helps NeoPixel
by moving the RMT interrupt to the second core as well.

When testing ESP32 I noticed that settings.toml writes won't apply
until after hard reset. This removes that constraint but still
requires the p [... cut]